ElvinC/gyroflow #3: Blackbox data import

Nikel9O, 2020-09-09:
Hi,
I've just stumbled upon [this](https://github.com/atomgomba/orangebox) project. 
I Think it could be useful to extract gyro data from black-box logs. 

what is your blackbox_extract.py expected to output exactly? might give it a look in the next few days.

ElvinC, 2020-09-09:
Hey,

The purpose of blackbox_extract.py is to take in a path to a .bbl file, and extract the timestamped gyro data. The exact format isn't that important since everything can be converted/scaled, but the format I've been using is an Nx4 numpy array, where each of the N rows consists of [time, gyroX,gyroY,gyroZ], where time is in seconds and rate in rad/s. Feel free to implement it if you want :)

Also, since orangebox and crisp (as suggested [here](https://github.com/ElvinC/gyroflow/issues/2)) are GPL licensed, the same has to be the case for this project for compatibility. I'll change it shortly.

Thanks

Nikel9O, 2020-09-24:
Hi,
i've finally made a flight with my new (used) hero 6. i thought this might be intresting/useful to you:
[image]
It's the shape of gopro gyro x data (extracted using your script) and blackbox gyro x data from the same flight.
they seem to be quite different

while z seems relatively similar:
[image]

ElvinC, 2020-09-24:
That's very interesting! Thanks a lot. I believe the coordinate system used for the gopro data is oriented differently compared to the standard betaflight one. The spikes in the blue gopro datapoints in your first graph seem to correspond the betaflight data in the second graph, just reversed. You can try to match those up to see how close they are. From a glance they do seem rather similar, so fingers crossed that it's close enough to be useful. The camera uptilt means that there's a coupling of yaw and roll in the unprocessed data, so I guess looking at the pitch axis gives the most fair comparison between the data.

I'll also try to make some further progress in the image processing code this weekend.

Nikel9O, 2020-09-25:
You're right! Seems that the data just needs to be rotated (gopro x):
[image]

the scheme between gopro gpmf and betaflight seems to be:
x => -y
y =>  z
z => -x

this is y:
[image]

and this is y accounting for a -25 degree rotation on the data (my camera angle):
[image]

I'd say they're pretty close!

ElvinC, 2020-09-25:
That's awesome and really encouraging. Nice work! Also, I think I'm getting close to having the gyro video synchronization stuff working.

b020, 2020-09-25:
Woa. Really nice job. I am subscribed to this project because I am really interested to see it working and use it on my weekend hobby flights footages. Unfortunately I am out of time to help with data and coding, since I am working a lot lately, I would have to learn python(currently I am a java dev), and I have a new born baby at home that drains out almost every spare minute I have left (lol).

Greetings from Brazil.
Keep up the good work, guys.

closed 2020-12-22 by ElvinC