ElvinC/gyroflow #6: "Flip gyro data" option

petitssoldats, 2020-11-18:
Hey mate, awesome work on this project!

I'm trying to to use `stabilizer.py` on a clip made with a 3d printed FPV drone frame (lot of oscillations due to resonance) and a flipped naked Hero 8. Here is the [video](https://www.dropbox.com/sh/ms89xk2y81ua8nb/AAD4entXa3IvDum-1EfNIlc0a?dl=0)

[image]

Like with RSGO, i've some weird oscillations in the frame_out but in RSGO the option `Flip gyro data` fix it.

Here's an easy way to implement the option? How can I do it ? 

Cheers

ElvinC, 2020-11-18:
Hey, thanks for the interest

`stabilizer.py` is not really ready to be used practically at the moment since I'm sorta using it as a sandbox for playing around with ideas for synchronization etc. so things will break constantly (you may have noticed all the random plots that pop up). if you look around line [404](https://github.com/ElvinC/gyroflow/blob/master/stabilizer.py#L404) I've been switching around the components of the angular velocity vector manually during testing. If you want you can try to mess around with those lines to match the orientation (I suspect the pitch and yaw values have to be negated). One of the next steps is to determine the orientation of an external logger automatically and including gyro orientation info in the camera calibrations to make it more useful.

petitssoldats, 2020-11-18:
Alright! I'll take a look, thanks! But I though you were using `stabilizer.py` as there is nothing [here](https://github.com/ElvinC/gyroflow/blob/master/gyroflow.py#L72).  Can I ask you to explain me your process to stab a video? Dont want to waste your time 😅

ElvinC, 2020-11-18:
Well, there's not really a process at the moment since a few things have to be implemented, namely a somewhat reliable method of doing sync and eventually a UI. The current sync implementation fails fairly often, especially when there's parallax. Regarding the files, the idea was to have the image processing/backend code in `stabilizer.py` and the UI stuff in `gyroflow.py`.

closed 2021-01-09 by ElvinC