ElvinC/gyroflow #7: Using VidStab for stabilization

alexagv, 2020-11-26:
For stabilization you could try to use the [VidStab](https://adamspannbauer.github.io/python_video_stab/html/usage.html#using-vidstab-class) library. The VidStab library is basically a two phase process for digital image stabilization, stage one is to estimate motion and step two is stabilizing. However since you already have the movement data (blackbox or GPMF) you can skip step one and parse them in stage two, as shown in their documentation. You would still need to integrate/convert the gyro values into relative motion in x, y and rotation and put them in a comma-separated file.

```
import numpy as np
from vidstab import VidStab

# Read in csv transform data, of form (delta x, delta y, delta angle):
transforms = np.loadtxt(TRANSFORMATIONS_PATH, delimiter=',')

# Create stabilizer and supply numpy array of transforms
stabilizer = VidStab()
stabilizer.transforms = transforms

# Apply stabilizing transforms to INPUT_VIDEO_PATH and save to OUTPUT_VIDEO_PATH
stabilizer.apply_transforms(INPUT_VIDEO_PATH, OUTPUT_VIDEO_PATH)
```

Good luck! Love the project idea, I have been thinking about the same for quite some time and might contribute if I get some time during the weekends 👍

ElvinC, 2020-11-26:
Thanks!
VidStab looks neat, but so far the main challenge has been with orientation calibration and syncing, so not sure how much it helps in that regards. I've seen some of your links already, so the current code is a bit of an amalgamation of different ideas. I'll probably try to convert some data to a format VirtualGimbal accepts and see how well it does. Their main repo is actively maintained and the results look excellent.

If you're interested in some more background material, I also found a master thesis with basically everything there is to know about the topic: https://www.research-collection.ethz.ch/handle/20.500.11850/155222

closed 2020-12-22 by ElvinC

bjin, 2021-02-20:
> I'll probably try to convert some data to a format VirtualGimbal accepts and see how well it does.

Is there any progress on this?