ElvinC/gyroflow #8: Any plans/guidance for adapting this to real-time applications?

harshilpatel312, 2020-12-22:


ElvinC, 2020-12-22:
That's currently not planned, although you can find a number of papers and videos on the topic (E.g. [Karpenko et. al.](https://graphics.stanford.edu/papers/stabilization/karpenko_gyro.pdf) and this [video](https://www.youtube.com/watch?v=yvMZa6y-6_M)). One of the main challenges is estimating the delay between the image data and the gyro data in real-time. Stabilizing in real time without a buffer also means that you only know the orientation leading up to the current orientation, so the stabilized can't "predict" the future motion.

I'm currently also working on a seperate hardware project to learn about orientation estimation etc, so it could be interesting to try out as a proof of concept, no guarantee though.

harshilpatel312, 2020-12-22:
Thanks for your prompt response. Both of those resources look promising.

You're right, time delay would certainly affect this. Motion estimation and the whole real-time thing may probably work better in situations with periodic/cyclic motions though.

closed 2021-01-09 by ElvinC