ElvinC/gyroflow #9: Queston about real time data pipeline

HenryKang1, 2020-12-28:
Hello. I am wondering that, does gopro provide real time sensor and video data?
It looks like people use the log and already dowloaded data.
I want to use Gopro for real time visual odometry purpose for mobile SLAM project.
If we can get the sensor data it will be really good. Thank you for reading it.

ElvinC, 2020-12-28:
Hey. I don't think GoPros support reading of sensor data in real time, although an alternative could be an externally mounted IMU. It should be possible to get live video data if the camera has an HDMI out.

HenryKang1, 2020-12-28:
Thank you for answering. I check that multiple youtube video said that gopro provide the real time video data using by ffmpeg and opencv. https://github.com/KonradIT/gopro-py-api 
In my personel experience, cheap wireless camera has the data packet  that includes extra information such as IMU data. However I am just planning to buy the gopro so I do not have gopro yet. I just wondering you have already have so many experience about IMU data in gopro. That is why I am asking did you try realtime processing which you grab wirelessly transported data and decoding the data and process it.

ElvinC, 2020-12-28:
I don't actually have a GoPro and have been relying on clips shared by others for testing purposes, so I haven't tried any realtime stuff. But good luck in getting it to work!

closed 2020-12-31 by HenryKang1