ElvinC/gyroflow #10: can import "from stabilizer import GPMFStabilizer"

krzysztofkuczek, 2021-01-06:
Hi,
Thank you for your awesome work !!!
I'm going to play with it, but some package is missing i think.
Which one

> (cv_cuda) PS C:\Temp\gyroflow-master\gyroflow-master> python.exe .\gyroflow.py
> Traceback (most recent call last):
>   File ".\gyroflow.py", line 13, in <module>
>     from stabilizer import GPMFStabilizer
>   File "C:\Temp\gyroflow-master\gyroflow-master\stabilizer.py", line 8, in <module>
>     from blackbox_extract import BlackboxExtractor
>   File "C:\Temp\gyroflow-master\gyroflow-master\blackbox_extract.py", line 1, in <module>
>     from orangebox import Parser
> ModuleNotFoundError: No module named 'orangebox'

Thank you in advance, 
Kris

ElvinC, 2021-01-06:
Orangebox is a blackbox log parser library, but note that full blackbox processing isn't finished yet and you have to rely on blackboxtogpmf for now. In general you can install libraries using pip, in this case `pip install orangebox`. If you're missing other packages, just install them in the same way. Some packages like cv2 have to be installed using a different name (opencv-python), but just search for it on google and you'll find it. If you get something working, please share the results here or in the discord server/FB group, that would be awesome.

krzysztofkuczek, 2021-01-06:
Thank you for your feedback. Installing orange box haven't change a lot
```
(cv_cuda) PS C:\Temp\gyroflow-master\gyroflow-master> pip install orangebox
Collecting orangebox
  Downloading orangebox-0.2.0-py3-none-any.whl (31 kB)
Installing collected packages: orangebox
Successfully installed orangebox-0.2.0
(cv_cuda) PS C:\Temp\gyroflow-master\gyroflow-master> python.exe .\gyroflow.py
Traceback (most recent call last):
  File ".\gyroflow.py", line 13, in <module>
    from stabilizer import GPMFStabilizer
  File "C:\Temp\gyroflow-master\gyroflow-master\stabilizer.py", line 9, in <module>
    from GPMF_gyro import Extractor
  File "C:\Temp\gyroflow-master\gyroflow-master\GPMF_gyro.py", line 4, in <module>
    import gpmf.parse as gpmf_parse
  File "C:\Temp\gyroflow-master\gyroflow-master\gpmf\parse.py", line 10, in <module>
    import construct
ModuleNotFoundError: No module named 'construct'
```

I'm python newbie, but have some experience with tensor flow and open cv, so a lot of (i think) required libs are already installed
```
(cv_cuda) PS C:\Temp\gyroflow-master\gyroflow-master> conda list
# packages in environment at C:\Users\kku\anaconda3\envs\cv_cuda:
#
# Name                    Version                   Build  Channel
_tflow_select             2.1.0                       gpu
absl-py                   0.11.0             pyhd3eb1b0_1
aiohttp                   3.6.3            py37he774522_0
astor                     0.8.1                    py37_0
async-timeout             3.0.1                    py37_0
attrs                     20.3.0             pyhd3eb1b0_0
blas                      1.0                         mkl
blinker                   1.4                      py37_0
brotlipy                  0.7.0           py37h2bbff1b_1003
ca-certificates           2020.12.5            h5b45459_0    conda-forge
cachetools                4.2.0              pyhd3eb1b0_0
certifi                   2020.12.5        py37h03978a9_0    conda-forge
cffi                      1.14.4           py37hcd4344a_0
chardet                   3.0.4           py37haa95532_1003
click                     7.1.2                      py_0
cryptography              3.3.1            py37hcd4344a_0
cudatoolkit               10.1.243             h74a9793_0
cudnn                     7.6.5                cuda10.1_0
freeglut                  3.2.1                h0e60522_0    conda-forge
freetype                  2.10.4               h546665d_0    conda-forge
gast                      0.2.2                    py37_0
google-auth               1.24.0             pyhd3eb1b0_0
google-auth-oauthlib      0.4.2              pyhd3eb1b0_2
google-pasta              0.2.0                      py_0
grpcio                    1.31.0           py37he7da953_0
h5py                      2.10.0           py37h5e291fa_0
hdf5                      1.10.4               h7ebc959_0
icc_rt                    2019.0.0             h0cc432a_1
icu                       67.1                 h33f27b4_0    conda-forge
idna                      2.10                       py_0
importlib-metadata        2.0.0                      py_1
intel-openmp              2020.2                      254
jasper                    2.0.14               h77af90b_2    conda-forge
jpeg                      9d                   h8ffe710_0    conda-forge
keras-applications        1.0.8                      py_1
keras-preprocessing       1.1.0                      py_1
libblas                   3.9.0           1_h8933c1f_netlib    conda-forge
libcblas                  3.9.0           3_hd5c7e75_netlib    conda-forge
libclang                  10.0.1          default_hf44288c_1    conda-forge
libiconv                  1.16                 he774522_0    conda-forge
liblapack                 3.9.0           3_hd5c7e75_netlib    conda-forge
liblapacke                3.9.0           3_hd5c7e75_netlib    conda-forge
libopencv                 4.5.0                    py37_6    conda-forge
libpng                    1.6.37               h1d00b33_2    conda-forge
[... 47 lines cut ...]
tensorboard-plugin-wit    1.6.0                      py_0
tensorflow                2.1.0           gpu_py37h7db9008_0
tensorflow-base           2.1.0           gpu_py37h55f5790_0
tensorflow-estimator      2.1.0              pyhd54b08b_0
tensorflow-gpu            2.1.0                h0d30ee6_0
termcolor                 1.1.0                    py37_1
urllib3                   1.26.2             pyhd3eb1b0_0
vc                        14.2                 h21ff451_1
vs2015_runtime            14.27.29016          h5e58377_2
werkzeug                  0.16.1                     py_0
wheel                     0.36.2             pyhd3eb1b0_0
win_inet_pton             1.1.0            py37haa95532_0
wincertstore              0.2                      py37_0
wrapt                     1.12.1           py37he774522_1
xz                        5.2.5                h62dcd97_1    conda-forge
yarl                      1.6.3            py37h2bbff1b_0
zipp                      3.4.0              pyhd3eb1b0_0
zlib                      1.2.11               h62dcd97_4
zstd                      1.4.8                h4e2f164_1    conda-forge
```

Of course i will share my test results.
I'm also looking some guide, how to check gyro log offset, and how exactly it should be done. 
I can imagine, that before flight there is need let say knock twice the camera, to shake it, then this patern in gyro data will be automatically detected by simple AI and matched with recorded video stream.... I going to try play with learning AI to that....
Kris

ElvinC, 2021-01-06:
Just keep installing any missing packages using pip. The updated gyro sync algorithm I've been working on seem to be able to match it within +/- 1 frame pretty much every time now, regardless of movement, so I don't think shaking it will be needed after all.

krzysztofkuczek, 2021-01-06:
Thank you it's working now!
Kris

closed 2021-01-06 by krzysztofkuczek

Abyl0109, 2022-05-11:
File "gyroflow.py", line 5, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
Hello, what i can do for solve this problem?

ElvinC, 2022-05-11:
@Abyl0109 This repo is legacy code, refer to https://github.com/gyroflow/gyroflow

Abyl0109, 2022-05-11:
I need to gitclone this link right?

>