ElvinC/gyroflow #15: export error

dentex, 2021-01-23:
Hello,
I've just cloned the repo and installed a few python packages with pip.
I wanted to try your code with a GoPro session 5 clip (I had to change the file extension from "MP4" to "mp4").

I set up the UI like this:
[image]

All the processing seems to complete successfully, but when I hit _export_ I have this error in the terminal (I also tried various combinations in the nearby checks and bitrate):

```
17:33:56 :: WriteGear :: ERROR :: BrokenPipeError caught, Wrong values passed to FFmpeg Pipe, Kindly Refer Docs!
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/vidgear/gears/writegear.py", line 315, in write
    self.__process.stdin.write(frame.tostring())
BrokenPipeError: [Errno 32] Pipe interrotta

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gyroflow.py", line 1519, in export_video
    self.stab.renderfile(start_time, stop_time, filename[0], out_size = out_size,
  File "/home/samuele/GIT/gyroflow/stabilizer.py", line 571, in renderfile
    out.write(concatted)
  File "/usr/local/lib/python3.8/dist-packages/vidgear/gears/writegear.py", line 321, in write
    raise ValueError  # for testing purpose only
ValueError
```

I'm on a Linux machine running Linux Mint 20.1.

```
~ $ ffmpeg 
ffmpeg version 4.3.git Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: 
  libavutil      56. 59.100 / 56. 59.100
  libavcodec     58.106.100 / 58.106.100
  libavformat    58. 56.100 / 58. 56.100
  libavdevice    58. 11.102 / 58. 11.102
  libavfilter     7. 87.100 /  7. 87.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
```

I hope this can be useful.
If any other info is needed, please let me know.

Thanks, bye.

ElvinC, 2021-01-23:
Not sure what could be causing the error. I'll try to take a look later, but in the meantime you can try this version which uses the standard OpenCV videowriter instead of vidgear https://github.com/ElvinC/gyroflow/tree/60d27aaafe15b17054f52c2f9786fde429281ff7

dentex, 2021-01-23:
Yes, it worked. 
https://www.dropbox.com/sh/7elhx8gtd8toi34/AACaa6V4kkCwkg9LAEoslqgja?dl=0 (there's a test video)
Thank you very much.

abhiTronix, 2021-01-25:
Hey @ElvinC, WriteGear automatically derives the frame-size for its `-s` flag from the first incoming frame:

https://github.com/abhiTronix/vidgear/blob/7018d166ddcdb5f585225b79cf87d0ef335cb550/vidgear/gears/writegear.py#L350-L360

therefore manually adding `-vf` flag for scaling may be interfering with it: 

https://github.com/ElvinC/gyroflow/blob/d814ddc80c979c01d76515231cf1c20a73cbcb22/stabilizer.py#L473

You can remove `-vf` flag since `-s` is internally mapped to `-vf=scale`, and they both kind of do the same work.

ElvinC, 2021-01-25:
@abhiTronix Thanks, I'll change that. Appreciate the help.

hyperion11, 2021-01-26:
have same problem
this version work
https://github.com/ElvinC/gyroflow/tree/60d27aaafe15b17054f52c2f9786fde429281ff7

abhiTronix, 2021-01-26:
@ElvinC Also, Sorry if I missed it earlier, `-pix_fmt` is assigned automatically in WriteGear:

https://github.com/abhiTronix/vidgear/blob/7018d166ddcdb5f585225b79cf87d0ef335cb550/vidgear/gears/writegear.py#L363-L370 

Please remove it.

ElvinC, 2021-01-26:
@abhiTronix That's noted. Thanks again.

alexagv, 2021-01-29:
Hi @dentex and @hyperion11 sorry for the issues, I never tested the VidGear on Linux, can check it out this weekend on a Ubuntu machine. Does any of you have a dedicated graphics card? Would be interesting to test HW encoding on Linux as well using FFmpeg Vaapi.

alexagv, 2021-01-29:
@abhiTronix Good catch!
Will add it to the branch with the fix for Linux as well.

hyperion11, 2021-01-29:
Hi @alexagv. Thanks for fix. I will check HW encoding! I have relatively old GTX 760.

alexagv, 2021-01-29:
> Hi @alexagv. Thanks for fix. I will check HW encoding! I have relatively old GTX 760.

If it doesnt work (without HW encoding), try to remove line 499 to 502 in stabilizer.py and see if that works. 
I guess you dont have FFmpeg compiled with libx264 enabled :) Will add a check if the encoders exist before using it

dentex, 2021-01-29:
Hi @alexagv thanks for taking a look.
I tried your version, but the problem persists, exporting with HW encoding or not (and also deleting the two lines you mentioned).
It's the same error message as in the 1st post up here.

(I have a Nvidia GTX 850M, on an Asus laptop)

abhiTronix, 2021-01-29:
@ElvinC @alexagv Thanks for including the purposed fix. Can you please put [`logging`](https://abhitronix.github.io/vidgear/latest/gears/writegear/non_compression/params/#logging) parameter to WriteGear at this line: 

https://github.com/ElvinC/gyroflow/blob/d814ddc80c979c01d76515231cf1c20a73cbcb22/stabilizer.py#L511

so that we can look at terminal log, and backtrack problem easily. 

@dentex You can directly put `logging = True` at above mentioned line to see debugged output on your terminal. Then you can paste that output here so we can see it too.

dentex, 2021-01-29:
I did a `git reset --hard master` to restore the original `stabilizer.py` at `d814ddc`

Then I added the line you said:
[image]

But there's no additional logging:

```17:28:55 :: WriteGear :: ERROR :: BrokenPipeError caught, Wrong values passed to FFmpeg Pipe, Kindly Refer Docs!
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/vidgear/gears/writegear.py", line 315, in write
    self.__process.stdin.write(frame.tostring())
BrokenPipeError: [Errno 32] Pipe interrotta

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "gyroflow.py", line 1519, in export_video
    self.stab.renderfile(start_time, stop_time, filename[0], out_size = out_size,
  File "/home/samuele/GIT/gyroflow/stabilizer.py", line 577, in renderfile
    out.write(frame_out)
  File "/usr/local/lib/python3.8/dist-packages/vidgear/gears/writegear.py", line 321, in write
    raise ValueError  # for testing purpose only
ValueError
```

abhiTronix, 2021-01-29:
@dentex No, Like I meant this `out = WriteGear(output_filename=outpath, logging=True, **output_params)`.

dentex, 2021-01-29:
oh yeah...

abhiTronix, 2021-01-29:
@dentex Apologies if I didn't explained it correctly.

dentex, 2021-01-29:
> @dentex Apologies if I didn't explained it correctly.

It's more about me being noob, don't worry. 

```Output file: /home/samuele/Scrivania/temp/GOPR0843_.mp4
17:38:30 :: WriteGear :: DEBUG :: Compression Mode is enabled therefore checking for valid FFmpeg executables.
17:38:30 :: WriteGear :: DEBUG :: Output Parameters: {'-input_framerate': 29.97002997002997, '-vf': 'scale=1920x1080', '-c:v': 'libx264', '-crf': '1', '-maxrate': '20.0M', '-bufsize': '24M', '-pix_fmt': 'yuv420p'}
17:38:30 :: Helper :: DEBUG :: Final FFmpeg Path: ffmpeg
17:38:30 :: Helper :: DEBUG :: FFmpeg validity Test Passed!
17:38:30 :: Helper :: DEBUG :: Found valid FFmpeg Version: `b'4.3.git'` installed on this system
17:38:30 :: WriteGear :: DEBUG :: Found valid FFmpeg executables: `ffmpeg`.
17:38:30 :: WriteGear :: DEBUG :: Compression Mode is configured properly!
FRAME: 0, IDX: 0
17:38:30 :: WriteGear :: DEBUG :: InputFrame => Height:1080 Width:1920 Channels:3
17:38:30 :: WriteGear :: DEBUG :: Setting Input framerate: 29.97002997002997
17:38:30 :: WriteGear :: DEBUG :: Executing FFmpeg command: `ffmpeg -y -f rawvideo -vcodec rawvideo -s 1920x1080 -pix_fmt bgr24 -framerate 29.97002997002997 -i - -vf scale=1920x1080 -c:v libx264 -crf 1 -maxrate 20.0M -bufsize 24M -pix_fmt yuv420p -vcodec libx264 -preset fast /home/samuele/Scrivania/temp/GOPR0843_.mp4`
ffmpeg version 4.3.git Copyright (c) 2000-2020 the FFmpeg developers
  built with gcc 7 (Ubuntu 7.5.0-3ubuntu1~18.04)
  configuration: 
  libavutil      56. 59.100 / 56. 59.100
  libavcodec     58.106.100 / 58.106.100
  libavformat    58. 56.100 / 58. 56.100
  libavdevice    58. 11.102 / 58. 11.102
  libavfilter     7. 87.100 /  7. 87.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
Unrecognized option 'crf'.
Error splitting the argument list: Option not found
17:38:30 :: WriteGear :: ERROR :: BrokenPipeError caught, Wrong values passed to FFmpeg Pipe, Kindly Refer Docs!
[...]
```
and then there's the part above

dentex, 2021-01-29:
`Unrecognized option 'crf'` in RED :)

abhiTronix, 2021-01-29:
@dentex Thank you so much. But why `crf` is culprit I wonder. 🤔

abhiTronix, 2021-01-29:
@dentex Please remove these two lines and then try: 

https://github.com/ElvinC/gyroflow/blob/d814ddc80c979c01d76515231cf1c20a73cbcb22/stabilizer.py#L504-L505

dentex, 2021-01-29:
I did and it logged this:

```
17:52:46 :: WriteGear :: DEBUG :: Executing FFmpeg command: `ffmpeg -y -f rawvideo -vcodec rawvideo -s 1920x1080 -pix_fmt bgr24 -framerate 29.97002997002997 -i - -crf 1 -maxrate 60.0M -bufsize 72M -pix_fmt yuv420p -vcodec libx264 -preset fast /home/samuele/Scrivania/temp/GOPR0843_stab.mp4`
ffmpeg version 4.3.git
[...]
Unrecognized option 'crf'.
```

SO I also removed `"-crf": "1"`, but somehow it derives that parameter somewhere else because then it logged:

```
17:55:32 :: WriteGear :: DEBUG :: Executing FFmpeg command: `ffmpeg -y -f rawvideo -vcodec rawvideo -s 1920x1080 -pix_fmt bgr24 -framerate 29.97002997002997 -i - -maxrate 60.0M -bufsize 72M -pix_fmt yuv420p -vcodec libx264 -crf 18 -preset fast /home/samuele/Scrivania/temp/GOPR0843_s.mp4`
ffmpeg version 4.3.git
[...]
Unrecognized option 'crf'.

abhiTronix, 2021-01-29:
@dentex Let's run bare minimum. Remove all these lines:

https://github.com/ElvinC/gyroflow/blob/d814ddc80c979c01d76515231cf1c20a73cbcb22/stabilizer.py#L504-L509

dentex, 2021-01-29:
```
18:07:17 :: WriteGear :: DEBUG :: Executing FFmpeg command: `ffmpeg -y -f rawvideo -vcodec rawvideo \
 -s 1920x1080 -pix_fmt bgr24 -framerate 29.97002997002997 -i - -vcodec libx264 \
 -crf 18 -preset fast /home/samuele/Scrivania/temp/GOPR0843_stabbed.mp4`
ffmpeg version 4.3.git
[...]
Unrecognized option 'crf'.
```

abhiTronix, 2021-01-29:
@dentex This is weird.

abhiTronix, 2021-01-29:
@dentex Ok can you paste the output of `ffmpeg -encoders`

abhiTronix, 2021-01-29:
@dentex Your installed FFmpeg build might lack `--enable-gpl --enable-libx264` which is required to use `-c:v libx264` a.k.a H264 encoder. More info [here](https://superuser.com/a/1303789)

dentex, 2021-01-29:
https://ghostbin.com/paste/AtHTp/ffmpeg

[ffmpeg]

abhiTronix, 2021-01-29:
> https://ghostbin.com/paste/AtHTp/ffmpeg
> 
> [ffmpeg]
> 

@dentex Yeah h264 encoder is clear missing. Hence That's why this error.

abhiTronix, 2021-01-29:
@dentex  Please use these commands to fix this error:

```sh
sudo add-apt-repository ppa:jonathonf/ffmpeg-4
sudo apt-get update && sudo apt install ffmpeg
```

Goodluck!

dentex, 2021-01-29:
@abhiTronix I try to avoid `ppa`s when I can, so I simply compiled latest `ffmpeg` from source
```
$ ffmpeg -buildconf
ffmpeg version N-100878-gbfdf03207b Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --enable-gpl --enable-libx264
  libavutil      56. 64.100 / 56. 64.100
  libavcodec     58.119.100 / 58.119.100
  libavformat    58. 65.101 / 58. 65.101
  libavdevice    58. 11.103 / 58. 11.103
  libavfilter     7.100.100 /  7.100.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100

  configuration:
    --enable-gpl
    --enable-libx264
```
...and indeed it does export the thing.
Thanks for your help.

abhiTronix, 2021-01-29:
> @abhiTronix I try to avoid ppas when I can, so I simply compiled latest ffmpeg from source

@dentex Understandable. Glad it helped.

@ElvinC this issue is resolved at this point. I'll advise to add a option to define custom ffmpeg encoder, so that user can use them according to their need.

closed 2021-01-29 by dentex

alexagv, 2021-01-29:
@abhiTronix 
Vidgear uses libx264 in compression mode, shouldn't it check that ffmpeg has libx264 enabled as well? 

Because now it will allow compression mode as long as ffmpeg is installed but it will fail if libx264 is not installed.
If it checked for both it could fallback on non-compression mode?
If i understand the footnote here correctly: https://abhitronix.github.io/vidgear/latest/gears/writegear/non_compression/overview/

abhiTronix, 2021-01-29:
> Vidgear uses libx264 in compression mode, shouldn't it check that ffmpeg has libx264 enabled as well?

@alexagv Yeah, I can add that feature. I'll look into it. Thanks.

alexagv, 2021-01-31:
> @ElvinC Also, Sorry if I missed it earlier, `-pix_fmt` is assigned automatically in WriteGear:
> 
> https://github.com/abhiTronix/vidgear/blob/7018d166ddcdb5f585225b79cf87d0ef335cb550/vidgear/gears/writegear.py#L363-L370
> 
> Please remove it.

Since we only use h264 encoding here there is no issue specifying `yuv420p`
I tried encoding with x264enc without `yuv420p` in the `output_params` and VidGear encoded a video that does not play in Quicktime etc. 

I see in the code yoi linked that yuv420p is not an option?

More info regarding on `yuv420p`:
https://trac.ffmpeg.org/wiki/Encode/H.264#Encodingfordumbplayers

alexagv, 2021-01-31:
@dentex try to install Nvidia encoder for ffmpeg and run this branch on your Linux machine and see if HW acceleration works for you:
https://github.com/alexagv/gyroflow/tree/vidgear-fix

abhiTronix, 2021-01-31:
> I tried encoding with x264enc without yuv420p in the output_params and VidGear encoded a video that does not play in Quicktime etc.

@alexagv It covers for current default H264 encoding, therefore any custom encoders like `x264enc` and supporting dumb players, you need to specify its requirements accordingly, and also specified in docs.

> I see in the code yoi linked that yuv420p is not an option?

`yuv420p` is not compatible directly with RAW OpenCV frames, that's why it is not specified.

dentex, 2021-01-31:
> ...try to install Nvidia encoder for ffmpeg...
Yes, this was not so straightforward :)

```
~/GIT/ffmpeg $ ffmpeg -buildconf
ffmpeg version N-100878-gbfdf03207b Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
  configuration: --enable-gpl --enable-libx264 --enable-nonfree --enable-nvenc
  libavutil      56. 64.100 / 56. 64.100
  libavcodec     58.119.100 / 58.119.100
  libavformat    58. 65.101 / 58. 65.101
  libavdevice    58. 11.103 / 58. 11.103
  libavfilter     7.100.100 /  7.100.100
  libswscale      5.  8.100 /  5.  8.100
  libswresample   3.  8.100 /  3.  8.100
  libpostproc    55.  8.100 / 55.  8.100

  configuration:
    --enable-gpl
    --enable-libx264
    --enable-nonfree
    --enable-nvenc
```

with HW encoding enabled:
```
WriteGear :: DEBUG :: Terminating WriteGear Processes.
frame=  299 fps= 30 q=7.0 Lsize=   72969kB time=00:00:09.94 bitrate=60116.8kbits/s speed=0.998x    
video:72966kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.003527%
```
vs HW encoding disabled:
```
WriteGear :: DEBUG :: Terminating WriteGear Processes.
frame=  299 fps= 12 q=-1.0 Lsize=   76439kB time=00:00:09.87 bitrate=63401.2kbits/s speed=0.386x    
video:76434kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.005707%
```

Full log at: https://ghostbin.com/paste/Bqsz2
(1st chunk is with HW encoding enabled, filename `test_HW-enc.mp4`
2nd part is without HW encoding, filename `test.mp4`)

alexagv, 2021-01-31:
> > ...try to install Nvidia encoder for ffmpeg...
> 
> Yes, this was not so straightforward :)
> 
> ```
> ~/GIT/ffmpeg $ ffmpeg -buildconf
> ffmpeg version N-100878-gbfdf03207b Copyright (c) 2000-2021 the FFmpeg developers
>   built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
>   configuration: --enable-gpl --enable-libx264 --enable-nonfree --enable-nvenc
>   libavutil      56. 64.100 / 56. 64.100
>   libavcodec     58.119.100 / 58.119.100
>   libavformat    58. 65.101 / 58. 65.101
>   libavdevice    58. 11.103 / 58. 11.103
>   libavfilter     7.100.100 /  7.100.100
>   libswscale      5.  8.100 /  5.  8.100
>   libswresample   3.  8.100 /  3.  8.100
>   libpostproc    55.  8.100 / 55.  8.100
> 
>   configuration:
>     --enable-gpl
>     --enable-libx264
>     --enable-nonfree
>     --enable-nvenc
> ```
> 
> I tried exporting the same section of a stabilized video with:
> 
> * HW encoding enabled:
> 
> ```
> WriteGear :: DEBUG :: Terminating WriteGear Processes.
> frame=  299 fps= 30 q=7.0 Lsize=   72969kB time=00:00:09.94 bitrate=60116.8kbits/s speed=0.998x    
> video:72966kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.003527%
> ```
> 
> * HW encoding disabled:
> 
> ```
> WriteGear :: DEBUG :: Terminating WriteGear Processes.
> frame=  299 fps= 12 q=-1.0 Lsize=   76439kB time=00:00:09.87 bitrate=63401.2kbits/s speed=0.386x    
> video:76434kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.005707%
> ```
> 
> Full log at: https://ghostbin.com/paste/Bqsz2
> 
> (1st chunk is with HW encoding enabled, filename `test_HW-enc.mp4`
> 2nd part is without HW encoding, filename `test.mp4`)

Hehe, I know 😅 
Thanks for testing! At least you got 3x faster export speeds with HW encoding 🚀

dentex, 2021-01-31:
> > > ...try to install Nvidia encoder for ffmpeg...
> > 
> > 
> > Yes, this was not so straightforward :)
> > ```
> > ~/GIT/ffmpeg $ ffmpeg -buildconf
> > ffmpeg version N-100878-gbfdf03207b Copyright (c) 2000-2021 the FFmpeg developers
> >   built with gcc 9 (Ubuntu 9.3.0-17ubuntu1~20.04)
> >   configuration: --enable-gpl --enable-libx264 --enable-nonfree --enable-nvenc
> >   libavutil      56. 64.100 / 56. 64.100
> >   libavcodec     58.119.100 / 58.119.100
> >   libavformat    58. 65.101 / 58. 65.101
> >   libavdevice    58. 11.103 / 58. 11.103
> >   libavfilter     7.100.100 /  7.100.100
> >   libswscale      5.  8.100 /  5.  8.100
> >   libswresample   3.  8.100 /  3.  8.100
> >   libpostproc    55.  8.100 / 55.  8.100
> > 
> >   configuration:
> >     --enable-gpl
> >     --enable-libx264
> >     --enable-nonfree
> >     --enable-nvenc
> > ```
> > 
> > 
> > I tried exporting the same section of a stabilized video with:
> > 
> > * HW encoding enabled:
> > 
> > ```
> > WriteGear :: DEBUG :: Terminating WriteGear Processes.
> > frame=  299 fps= 30 q=7.0 Lsize=   72969kB time=00:00:09.94 bitrate=60116.8kbits/s speed=0.998x    
> > video:72966kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.003527%
> > ```
> > 
> > 
> > 
> > * HW encoding disabled:
> > 
> > ```
> > WriteGear :: DEBUG :: Terminating WriteGear Processes.
> > frame=  299 fps= 12 q=-1.0 Lsize=   76439kB time=00:00:09.87 bitrate=63401.2kbits/s speed=0.386x    
> > video:76434kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.005707%
> > ```
> > 
> > 
> > Full log at: https://ghostbin.com/paste/Bqsz2
> > (1st chunk is with HW encoding enabled, filename `test_HW-enc.mp4`
> > 2nd part is without HW encoding, filename `test.mp4`)
> 
> Hehe, I know sweat_smile
> Thanks for testing! At least you got 3x faster export speeds with HW encoding rocket

You're more than welcome.
I love what you guys are doing... I'll try to help whenever I can.