ElvinC/gyroflow #20: Feature request: option to only apply lens correction, no stabilization

avsaase, 2021-02-09:
Basically the title. I think the lens correction works great, and I would like to render a video that corrects the distortion without adding any stabilization. Right now Gyroflow assumes not providing a bblog means the gyro data is embedded in the video.

Thanks!

ElvinC, 2021-02-09:
I'll note it down somewhere. In the meantime you can also add a random gyro log, use low value for number of analysis frames, let it sync to a random place with smoothness to zero and export the undistorted result.

avsaase, 2021-02-09:
I'll try that. And thanks again for your amazing work.

kafie1980, 2021-04-12:
The lens correction is working well and the camera calibration is easy to work with to setup a camera lens. 

I would like to add my vote towards the request to be able to run lens correction without going through the video stabilizer. 

Similar to how non-linear stretch is an independent process.

IsaacRivasCayuela, 2021-10-06:
If you are interested you can already correct any fisheye/curved perspective (without stabilising) with this robust plugin for Davinci Resolve (paid version): https://github.com/roukou3/DCTL
It's the Ptgui Dctl file + defisheye.h that has to be copied in the LUT folder, it takes some learning -in Davinci Resolve 16, 17 it has to be done some workarounds to work properly in 4:3 footage- but it's a very polished tool and very fast for rendering; any help just ask me or the author in his Github. I use it for my pre-stabilised GoPro footage: https://youtu.be/WuJv0gcKcQs
ElvinC maybe you'll find it interesting as it can work in combination with your Gyroflow, for example I intend to use Gyroflow with a wide angle that's not properly centred in the sensor so maybe a manual lens calibration versus an automatic has its advantages where I can fine tune the asymmetry, rectilinear errors, animate with keyframes, etc after being stabilised in a 3D curved projection with Gyroflow.

Cheers