ElvinC/gyroflow #22: Error when using INAV blackbox log decode by blackbox tools

avsaase, 2021-02-11:
Not sure if this is a bug, or that INAV blackbox logs are just not supported. If the latter please considere this a feature request ;).

With that out of the way, I am trying to use an INAV blackbox log to stabilize some video. INAV blackbox logs are `.txt` files, so I converted it to `.csv` using [cleanflight/blackbox-tools](https://github.com/cleanflight/blackbox-tools). When trying to run the synchronization Gyroflow throws the error  `ValueError: 'gyroADC[0]' is not in list`. `gyroADC[0]` is in the csv file though. Any help would be much appreciated. Thanks!

ElvinC, 2021-02-11:
If you could send the CSV file you created I can take a look

avsaase, 2021-02-11:
Sorry forgot to attach then. Here they are: [blackbox_log_2021-02-11_214621.zip](https://github.com/ElvinC/gyroflow/files/5968600/blackbox_log_2021-02-11_214621.zip)

ElvinC, 2021-02-11:
Oh I see what's happening. The csv file generated by blackbox-tools doesn't add citation marks to strings:
`loopIteration, time (us), axisRate[0], axisRate[1], axisRate[2], axisP[0]`
While the output of the blackbox explorer gives:
`"loopIteration","time","axisP[0]","axisP[1]","axisP[2]","axisI[0]"`
And `"gyroADC[0]"` isn't found since the item in the list is `" gyroADC[0]"` with a space. I tried opening the TXT file in the blackbox explorer and the exported csv works fine.

avsaase, 2021-02-11:
Is this with the Betaflight blackbox explorer or the INAV version. Could be the same of course but I cannot try it right now.

ElvinC, 2021-02-11:
I was using the betaflight one but yeah, not sure if there's a difference.

avsaase, 2021-02-11:
Thanks a lot. I'll try the csv export from the INAV blackbox explorer tomorrow. It would be nice if the `.txt` were directly supported in a future version. Not sure how much work would go into that.

closed 2021-02-23 by ElvinC