ElvinC/gyroflow #28: No more stabilised data, unplayable video

SebastianKumor, 2021-03-03:
Hello, i was testing gyroflow and all went well but when the stabilization was at 98.5% i got `No more stabilization data` and I was unable to play the outputted generated file. Then I tried again and setting the video stop to 10 seconds less and the stabilisation process froze on 99.9 percent and the output file was unplayable as well. For some reason I cannot get playable video file. I am running it on mac. 

[image]

ElvinC, 2021-03-06:
Thanks for the info about what encoders work. The progress percent is not really "stuck" since it doesn't print a message after completion (sorry about that) `no more stabilization data` should also terminate and save the file, so it was a problem with the encoder.

closed 2021-04-17 by ElvinC

petervieth, 2021-04-25:
So Sebastian did not get this working, so maybe a bit early to close the bug? I have exactly the same error on a 2015 Macbook, also tried same options as Sebastian, and behavior is the same: gyroflow runs but the output doesn't open in Quicktime.  Davinci Resolve shows the media as "offline" when imported.  I was, however, able to open it in VLC.

ElvinC, 2021-04-25:
@petervieth sorry, forgot to add a message here. If you're using the executable, then the bug should have been fixed on the master branch. Let me know if you're still having problems using that. New executables will be out soonish.

petervieth, 2021-04-25:
Thanks for the explanation. Unfortunately I don't use python and getting it working on Mac is not fun (I just tried again for half an hour and gave up). I know clobbering python 2.7 is bad.

ElvinC, 2021-04-26:
@petervieth Did you run `poetry install` first, followed by `poetry run python gyroflow.py`? I just tried with a fresh clone of the master branch and didn't have any issues. Can you send the the error messages you're getting?

petervieth, 2021-04-26:
Yes, I went to the poetry site and followed the instructions. I can't seem to figure out how to get it to save the environment once I close the shell though (I just ran it using $HOME/.poetry/bin/poetry )

On Windows, the error is: Unable to find installation candidates for opencv-python (4.1.2.30)
On MacOS: NoCompatiblePythonVersionFound
But I installed python 3.7 so I know that's not true (pyenv install 3.7.0), it probably just can't find it and I'm not sure how to tell it where it is, or where 3.7 lives.

petervieth, 2021-04-29:
It's because I was using Python 3.9 and you're using something earlier. There is no 4.1.2 of opencv-python for 3.9 of any architecture, which is why poetry can't find it.  I figured this out by trying to use a local repo.  Maybe you wish to add a note about this to the webpage for python n00bs like myself :)

ElvinC, 2021-04-29:
@petervieth oooh that makes sense, nice find. I'll make sure to add a note. Thanks!