ElvinC/gyroflow #31: Feature Request: Ubuntu Packaging

Schlaegel, 2021-03-08:
I see there is a placeholder for Ubuntu packaging on the website. I tried in vain to follow the install, and after working around a few dependency errors, I finally gave up.

If you could provide either a deb or snap, it would be awesome.

Thanks for the (free) software, from what I see on the video and website, it looks great.

I would offer to help with the packaging, but I was never able to get my own copy working, and I'm new to Ubuntu, after decades on some form of Redhat/Fedora.

ElvinC, 2021-03-08:
Oh sorry about that, I was planning to look into it but haven't had a chance to do so. I had a quick try at packaging it in a VM but had a couple of problems, will have to look more into it. In the meantime can you try the following to see if it works?

1. Navigate to some folder
2. Run `git clone https://github.com/ElvinC/gyroflow.git`
3. Install poetry as described here: https://python-poetry.org/docs/
4. Navigate to the gyroflow folder and run `poetry install` to install dependencies
5. Install ffmpeg using `sudo apt install ffmpeg`
6. Run using `poetry run python gyroflow.py`

ElvinC, 2021-03-09:
Did you navigate to the gyroflow folder containing `gyroflow.py` before running `poetry run python gyroflow.py`. I get the same error if I run the command from a random folder.

Schlaegel, 2021-03-11:
Okay that was it. When installing the dependencies, I must have left the gyroflow directory. So, my system is now running, but in the long run, it would still be great if you or someone else more knowledgeable than myself could make a snap or a deb.

ElvinC, 2021-08-30:
Managed to set up a ubuntu VM and pack gyroflow 0.3.0. See newest release

closed 2021-08-30 by ElvinC