ElvinC/gyroflow #32: Blank output video, black rendering screen

DarthPlasma, 2021-03-09:
I encountered an issue with the GUI version of Gyroflow.

After proper sinchronization was achieved I was not able to render my video, despite no error given.

ElvinC, 2021-03-09:
Please provide a bit more information. 
* What system are you on?
* What camera are you using?
* Does it begin the rendering process at all, if so, is the preview black?
* If a video file is saved, what's the filesize?
* Can it be opened with a video player or does it give an error? 
* Did you use a newly made lens preset or a premade one? Try a premade one if you're using a new one.

DarthPlasma, 2021-03-09:
**What system are you on?**
I'm using a Windows 10 PC, Core I7 6670, 16Gb Ram, Geforce GTX 1060 6gb. 
Running the standalone exe, not through python.

**What camera are you using?**
DJI Osmo Action (calibrated first with 10 frames, If needed I can upload the profile for other users), running at 4K 60fps cinelike

**Does it begin the rendering process at all, if so, is the preview black?**
It starts rendering and the progress goes on correctly, but the rendering screen is black (except for the debug text in topleft corner). After the rendering is completed the final file plays but it is black or has only the debug text if checked.

**If a video file is saved, what's the filesize?**
Actually is very small, something around 30megabytes (as it is totally black this does not surprise me)

**Can it be opened with a video player or does it give an error?**
The video gets opened, and displays the debug text if checked

**Did you use a newly made lens preset or a premade one? Try a premade one if you're using a new one.**
I made a lens preset, but i will try.

DarthPlasma, 2021-03-11:
You were right! It was a lens calibration issue. Now the video renders perfectly and with some frame fine tuning sinchronization it looks very smooth and nice.

If you need I can upload the DJI Osmo preset (4K 16:9 Linearized). Next time I will try to with 4K 4:3 not linearized just to have a little more fov and cropping margin.

Thank you!

ElvinC, 2021-03-11:
That would be awesome. You can send it using the form here: https://elvinchen.com/gyroflow/lens/

closed 2021-04-17 by ElvinC