ElvinC/gyroflow #33: Feature request: Support ardupilot logs

hyperion11, 2021-03-09:
Hi! I found tool to convert ardupilot logs to CSV format
Here example log file:
https://1drv.ms/u/s!AuFPVI_eDaZbhY4X3UuhfYfvd-SgFg?e=ox3Lfe
i think IMU_GyrX,IMU_GyrY,IMU_GyrZ already in radians
-0.0003537596 | -0.0006091184 | 0.0002790984
It would be great to add ardupilot support to the program!

ElvinC, 2021-03-09:
Nice, I'll take a look but that should be fairly easy to add. Do you know how XYZ maps to roll/pitch/yaw? And since people have done all sort of crazy builds with ardupilot, can I assume for now that the camera is mounted like a typical FPV drone at an angle?

hyperion11, 2021-03-10:
i will check XYZ map and answer today. I think left camera angle as is in BF log analyzer. In my case on plane dart250g there is about zero angle. But on quad it may be different, so i think let user calculate angle of camera and insert this value to program

hyperion11, 2021-03-10:
gx - roll
gy - pitch
gz - yaw

didokiuchukov, 2021-04-19:
I have several flights on fixed wing, recorded with DJI fpv onboard video and good logs from Arduplane/F705WSE. Can test.

ElvinC, 2021-04-19:
@didokiuchukov Could you send me a log and matching video clip I can use for testing? That would be awesome

didokiuchukov, 2021-04-21:
Hi Sure,

You have preferred file service or upload anywhere and share the link?

Best,

ElvinC, 2021-04-21:
It doesn't really matter, easiest is to upload somewhere and send a link.

ElvinC, 2021-07-26:
Closing this due to 50hz logging rate. At least 100 Hz of clean data is needed. Feel free to update if there's more info.

TMac54, 2021-11-04:
> Closing this due to 50hz logging rate. At least 100 Hz of clean data is needed. Feel free to update if there's more info.
@hyperion11
On 4 Nov 2021, I asked ArduPilot about possibly increasing logging rate to support gyroflow.  Here's the answer i got from Peter (Dev Team): "Peter Hall — Today at 12:39 PM
You can use the rate message, turn on fast attitude logging and it will be 400hz

ElvinC, 2021-11-04:
Thanks for the info. If anyone has an 400 Hz ardupilot log available, feel free to send it here.

TMac54, 2021-11-04:
> Hi! I found tool to convert ardupilot logs to CSV format. In mission planner, CTRL+F -> Dashware -> Select .bin log file -> fill only IMU in text field.
> 
> Here example converted log file: https://1drv.ms/u/s!AuFPVI_eDaZbhY4X3UuhfYfvd-SgFg?e=ox3Lfe
> 
> i think IMU_GyrX,IMU_GyrY,IMU_GyrZ already in radians -0.0003537596 | -0.0006091184 | 0.0002790984 It would be great to add ardupilot support to the program!

"You can use the rate message, turn on fast attitude logging and it will be 400hz". Peter Hall (ArduPilot Dev Team) has confirmed the way you set 400 Hz Attitude Logging by going to Full Parameter List in Mission Planner. Search for LOG_BITMASK. Toggle on ATTITUDE_FAST rather than default of ATTITUDE_MED.   You'll notice in the attached image, there's also an IMU_FAST toggle box.  I'm not sure which would work best for Gyroflow, but I mention this since you spoke of "IMU" above.
[image]

TMac54, 2021-11-05:
> Thanks for the info. If anyone has a 400 Hz ardupilot log available, feel free to send it here.

OK, Elvin, here's what I have done.  I have been in communication on Discord ArduPilot site with Peter Hall.  As mentioned above, he has stated to turn on ATTTIUDE_FAST and the logging rate will be 400 Hz.  I have turned on ATTITUDE_FAST and logged a short flight.  I have then used the method described above by @hyperion11 (CTRL+F -> Dashware -> Select .bin log file -> fill only RATE in text field).  Notice here, I have used "RATE" instead of IMU.  RATE is what was recommended by Peter Hall in ArduPilot Discord site:

"Peter Hall — Today at 8:55 AM (Nov5, 2021)
Filtered gryo data is in RATE. You could use one of the IMU messages, i'm not sure off the top of my head how fast they are, but then you get raw data loosing the low pass / notch ect filtering that AP does."

So, I have attached [here the CSV file](https://drive.google.com/file/d/1JFcwo-TcCLuF92WRZw4-SeRK78PrB0wz/view?usp=sharing) for the RATE log info supposedly captured at 400 Hz.  Hopefully, this is the info you need at the correct rate for Gyroflow.  If there is anything else you need that I am able to provide, I would be happy to do so.  However, yo may also find it more useful to work directly with the Dev's on ArduPilot Discord site as they are obviously much more knowledgeable and experienced with ArduPilot than I (I've just started with it!).  Please let mew know if this information is helpful for you to proceed and if there's anything else I can do for you .  Thank you for all yo do!

skorokithakis, 2021-11-06:
@ElvinC is the high logging rate still required for planes? Most of the vibration is low-frequency wind roll.

TMac54, 2021-11-08:
> Thanks for the info. If anyone has a 400 Hz ardupilot log available, feel free to send it here.

From Peter Hall ArduPilot Dev) in ArduPilot Discord
"Peter Hall — 11/05/2021
If it works OK we could look at adding a dedicated Gyroflow log message to output exactly what is needed. Then you can have just the stuff you need without having huge logs due to all the other stuff that is in the existing logs."

TMac54, 2021-11-23:
@ElvinC I understand completely you're busy.  Just wondering if you have any plans for Gyroflow to support ArduPilot logs as discussed above.  If so, is there any other information you would like or need to make this happen that I might be able to provide you with?  Thanks again for all your gyroflow efforts. Hope your studies are progressing well!

ElvinC, 2021-11-24:
Thanks for the reminder. I added the parser to the dev branch assuming it has the same xyz orientation as the betaflight blackbox, which may or may not be the case. If anyone knows the x y z reference of the ardupilot logs, that would be great. For reference gyroflow uses this: 
[image]
Alternatively a short handheld clip with isolated pitch, roll, yaw movements + ardupilot log would be very useful

skorokithakis, 2021-11-24:
@ElvinC I can provide this no problem, I have an FC sitting on my desk. It'll have to be on Friday, though.

hyperion11, 2021-11-24:
https://1drv.ms/u/s!AuFPVI_eDaZbhsVAZBQe6ZDNZKIX9g?e=VmgwEd
hi, here logs+csv+video from gopro7

ElvinC, 2021-11-25:
Alright the dev branch code has been updated and should now work correctly with high sample rate logs. The sent provided log/video combination was at a low sampling rate (<50 Hz), so the output is a bit jumpy, though I'm quite surprised it worked at all:
https://user-images.githubusercontent.com/20195216/143507259-1c388b88-1585-4cdf-8d25-5dee3db6fe32.MP4

closed 2021-11-25 by ElvinC

ElvinC, 2021-12-03:
Turned out that a few other things had to be changed for the 400 Hz logs to work. This has been added. See https://github.com/ArduPilot/ardupilot/issues/19376 for more info