ElvinC/gyroflow #35: Can't stabilize videos on Debian 10

nik012003, 2021-03-14:
I'm running Debian 10 with KDE.

nuxnik, 2021-03-19:
I also had this problem in Fedora using dwm window manager. I changed the pyproject.toml to:

`
numpy = "^1.19.3"
opencv-python = "~4.5.1"
`
then ran

`poetry install`

Then I removed the following file: /home/USERNAME/.cache/pypoetry/virtualenvs/gyroflow-XXX-py3.9/lib64/python3.9/site-packages/cv2/qt/plugins/platforms/libqxcb.so

ElvinC, 2021-03-20:
@nuxnik Thanks for the info

Dave4675, 2021-07-31:
Hi, I'm using Ubuntu 20.04.2 and I followed elvin chen intructions to get gyroflow to work, but I get the same error when I try to export the file, gyroflow itself has some bugs but it works fine. I can't urdenstand why the software itself works good and crashes when I try to export a hopefully stabilized video

krawa, 2021-08-04:
Ubuntu 20.04. Same error. I got the export when I unchecked the "Display preview during rendering"