ElvinC/gyroflow #37: can we do it Real time on Nvidia Jetson Boards

gyillikci, 2021-04-01:
Hi guys amazing job 👍 

I was wondering if we feed gyro data and video feed Jetson Nano can handle realtime manipulation of the video so that we can get a clean video. I am considering it to use as fpv stabilizer to use in OpenHD https://github.com/OpenHD/Open.HD

Thanks

ElvinC, 2021-04-01:
Hey. The topic of realtime has been brought up previously. We know it works from smartphones, gopros etc, but it has a couple of challenges and disadvantages. One of the challenges is achieving accurate sync between video and gyro in realtime, although there's a bunch of literature on the topic. This could also be an issue if the delay between the two varies over time. Regarding the quality of the output, one disadvantage with realtime stabilization is the lack of the "predictive" movements possible in offline methods, although buffering a few frames should give something similar.

I've got an IMU test board with USB, so I might try something with a webcam and manual delay to see what happens.

Could you describe what sort of setup you're thinking of? Is the processing done onboard or offloaded to a groundstation? How is the gyro data captured and transferred?.

Also, VirtualGimbal (another gyro-based image stabilization project) has an optimized real time version. Although I haven't tried it myself, it seems to work fairly well. Take a look:
https://www.youtube.com/watch?v=ft6v7h5kN6g & https://github.com/yossato/virtualgimbal_ros

gyillikci, 2021-04-01:
So my approach would be get the video with csi camera on jetson nx or nano and read the gyro data via serial to usb converter and  stream in to the Opencv or Visionworks based realtime video processing. since csi raspberry pi v2 camera can stream 60 fps with very low latency I would first just translate the image corresponding to rapid changes (fft may be required) and make opposite displacement on the image against gyro read. I would not do any image warping or perspective correction. Just want to see If the video would get little bit more stabilized.  
even simpler approach could be when you vibrate the imu expect to see vibration on the image stream. If you get satisfied with the time alignment then you just move the image frame opposite of the amplitude coming from imu in deltaTime. and the delta time is around 33 ms for 60 fps.

caruofc, 2022-10-30:
> So my approach would be get the video with csi camera on jetson nx or nano and read the gyro data via serial to usb converter and stream in to the Opencv or Visionworks based realtime video [...]

Hi @gyillikci, I am trying to implement similar thing. Have you found a solution? How effective it is? Would appreciate if you could share your experience.

gyillikci, 2022-10-30:
> 

I ll be doing it in couple of weeks. let you know from here.

caruofc, 2022-10-31:
> > 
> 
> I ll be doing it in couple of weeks. let you know from here.

Great! would love to know the outcome. Meanwhile, I found this: https://www.mdpi.com/1424-8220/21/18/6219, couldn't find any implementation though.