ElvinC/gyroflow #38: Feature request: Custom background color

Amzd, 2021-04-02:
I know the black parts aren't really a background color but just empty space but it would be cool if we could fill it with a custom color. 

This would make it nicer to post on places where you control the background color like on your own website. 

Up until we get automatic zooming this will make it a little bit nicer to watch.

truog2003, 2021-04-02:
Cool !...

ElvinC, 2021-04-02:
Good idea. Other than background color selection, there are also different ways the empty area can be filled in. Some alternatives are reflection and continue with last color:
[image]
[image]

I'll make sure to add these options

Amzd, 2021-04-02:
That reflection actually properly tricks my brain 🤯

ElvinC, 2021-06-18:
Added in dev branch

closed 2021-06-18 by ElvinC