ElvinC/gyroflow #42: Option to use OpenTX telemetry data log as a Gyro log substitute for stabilization

kafie1980, 2021-04-20:
Problem: Most micro and whoop boards lack an onboard blackbox recording option and adding an external Openlager/OpenLogger type device is out of the possible options as most FC's on whoops have 2 or less Uarts and the percentage weight increase is huge in case of micros/whoops. 

Proposed workaround: Possibility of using OpenTX telemetry data log as a substitute for a Blackbox Gyro log for video stabilization.

avsaase, 2021-04-20:
I think you need at least a 100Hz logging rate to use for video stabilization, probably more. I don't think OpenTX can log that fast.

ElvinC, 2021-04-20:
Yep, it _might_ work down to 50 Hz if properly filtered etc, but at least 100 Hz is preferred.

kafie1980, 2021-04-20:
I guess this idea is not really worth the effort because I doubt OpenTX can log telemetry more 10hz as 0.1 seconds is the lowest value we can set for the SD logs function. 

I am assuming there is not enough processing power on OpenTX transmitters and probably the rate at which telemetry arrives at the transmitter limits this to be of enough resolution/detail.

Unless we interpolate the values from 10 hz to say 50 or 100 hz but then again as you all have explained there will not be enough detail or resolution in the data to stabilize the video effectively.

closed 2021-04-28 by kafie1980