ElvinC/gyroflow #45: Add Gopro Hero 8 Black lens preset (Attached .json file)

kafie1980, 2021-05-08:
I have created a lens preset file for the Gopro Hero 8 Black and had submitted it via the Gyroflow lens preset form ( https://forms.gle/6iKrUQWqCFxVghLd6 )about a week ago but it has not been added to the presets list yet.

My Sony_Action_Cam_HDR-AS30V preset was uploaded much faster and so I assume the dev's are busy.

Since the Gopro 8 is a popular camera and there is no official preset available, I am attaching the preset I created here in case anyone needs it before we have an official one added to the repository.

Please remove the .txt extension as I am unable to upload .json files to this message.

[GoPro_Hero_8_Black_2_7K_4by3_60fps_wide.json.txt](https://github.com/ElvinC/gyroflow/files/6445200/GoPro_Hero_8_Black_2_7K_4by3_60fps_wide.json.txt)

ElvinC, 2021-05-08:
Sorry about that, will add now. Thanks!

kafie1980, 2021-05-08:
Awesome!!!

closed 2021-05-08 by ElvinC