ElvinC/gyroflow #48: Ignore training/leading whitespace in csv headers

maxbl4, 2021-05-22:
INav blackbox converter adds whitespace to format CSV. I have to manually remove whitespace from headers in order for gyroflow to accept logs

ElvinC, 2021-05-22:
Could you attach an INAV log I can take a look at?

avsaase, 2021-05-22:
@ElvinC this is the same issue I reported in https://github.com/ElvinC/gyroflow/issues/22.

ElvinC, 2021-05-22:
Ah I see, I've added a fix on my dev version

closed 2021-05-31 by maxbl4

OMGChase, 2021-08-05:
> Ah I see, I've added a fix on my dev version. Maybe someone can also mention it to the INav blackbox converter people, unless the whitespace is on purpose.

Hello. Brought it up here with iNav guys and has been fixed in a pull request. Can see here:
https://github.com/iNavFlight/blackbox-tools/issues/40