ElvinC/gyroflow #49: Add configurable limit to screen movement

maxbl4, 2021-05-22:
Right now on sharp moves video may be moved too much, so black borders will be seen at any crop. Please add configurable limit for video movement

ElvinC, 2021-05-22:
Something like this is currently being tried out, and marcroe has a branch with an alternative orientation filtering method: https://github.com/marcroe/gyroflow/tree/nonlinearfilter

Feel free to join the discord linked in the readme if you want to try it out and give feedback.

maxbl4, 2021-05-31:
I've tried the latest dev version. Works great most of the time. I would only want to be able to set the limit on adaptive zoom. Right now in case of flip or roll it zooms in to 100500% :)

ElvinC, 2021-07-26:
Basic rotation limit added to dev branch. More options on the way

closed 2021-07-26 by ElvinC