ElvinC/gyroflow #55: RuntimeError

Erik6209, 2021-06-13:
Hi, i keep getting a runtime error while trying to "apply settings and compute sync".

[image]
[image]
[image]

does anyone know what i'm doing wrong ?

ElvinC, 2021-06-13:
It seems like the blackbox parser used sometimes doesn't parse the logs correctly (maybe if it's slightly corrupted), even though the blackbox explorer works fine. You can open the log in the blackbox explorer, export a CSV file, and use that instead of the raw log. (In the dev version I've added an error message when that happens explaining what to do)

Erik6209, 2021-06-14:
@ElvinC that worked for me, thank you !
Overall the video looks great stabilized, but in some moments it wobbles. Is that caused by unclean gyro-data ?

ElvinC, 2021-06-18:
> Is that caused by unclean gyro-data

Could be that or the sync

closed 2021-06-18 by ElvinC