ElvinC/gyroflow #56: Updated opencv-python does not run

MiniGod, 2021-06-14:
On the `dev` branch, I had to downgrade `opencv-python` to run gyroflow.

This version doesn't work: `4.5.1.48` (from lock file)
This version works: `4.1.2.30` (changed in 4a5237fe)

I'm using Ubuntu 20.04.
The error with `opencv-python@4.5.1.48`:
```
$ poetry run python gyroflow.py                                                                            
QObject::moveToThread: Current thread (0x2b88af0) is not the object's thread (0x2c38780).                                                                                                                            
Cannot move to target thread (0x2b88af0)                                                                  
                                                     
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/home/kristjan/.cache/pypoetry/virtualenvs/gyroflow-oJ6C59RA-py3.8/lib/python3.8/site-packages/cv2/qt/plugins" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
                                                     
Available platform plugins are: xcb, eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, webgl.                                               
                                                                                                                                                                                                                     
Aborted (core dumped)
```

ElvinC, 2021-06-14:
Thanks for letting me know. The reason for that change was due to some newer python versions not supporting old opencv versions. I'll have a look into it. If there's no nice solution I guess the python version can also be further constrained.

erikgoldenstein, 2021-06-14:
I had the same Problem,
I have made a pull-request containing a fix for newer opencv versions.
#58

ElvinC, 2021-07-23:
Issue still seems to persist for some ubuntu users. Will look more into it

ElvinC, 2021-08-30:
Based on some cross platform testing this should be fixed now

closed 2021-08-30 by ElvinC