ElvinC/gyroflow #62: Export stopping when overwriting input file

Gro2mi, 2021-07-24:
When you choose your input filename as output filename you get warned about overwriting the file. If you accept nonetheless the file gets overwritten with the new partial file and stops exporting.

## Console output
```
Output file: D:/git/FPV/videos/GH011159.MP4
Starting to compute optimal Fov
QCoreApplication::exec: The event loop is already running
Done computing optimal Fov
frame: 579, 0/1540 (0.0%)
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 1, offset 0x9b3c22d: partial file
frame: 584, 5/1540 (0.3%)
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 1, offset 0x9b3c439: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 0, offset 0x9b3c637: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 0, offset 0x9b75de0: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 1, offset 0x9bac8b7: partial file
[mov,mp4,m4a,3gp,3g2,mj2 @ 0000026ecce18080] stream 0, offset 0x9bacac1: partial file
...
```

ElvinC, 2021-07-24:
Good catch. I'll add a check to disallow overwriting the input file

ElvinC, 2021-07-26:
Fix in dev

closed 2021-07-26 by ElvinC