ElvinC/gyroflow #73: Some windows won't disappear/appear at some point when not needed

artem-alexandrov, 2021-08-12:
Environment:
Mac OS X 11.4
gyroflow 0.3.0-dev

STR:
1. Launch the app
2. Open fancy stabilizer
3. Select input file, camera preset, gyro data
4. Open gyro log type selector (this window will later appear) and choose whatever you need (RC5 csv in my case)
5. Go to sync/stab tab and press "Apply settings and compute sync"

AR:
No unneeded windows appear on screen

ER:
Gyro log type selector appears on screen

Workaround here is to go back to "Input" tab and open/close that selector again. But this is quite annoying.
The same happens with any windows you've opened before, including "Save" dialog (with selecting where to save exported video

Screenshots of the problem:
[image]

artem-alexandrov, 2021-08-12:
Update:

This happens when `plt.show()` is executed from `stabilizer.py` in line 253. 
Is it intended to be called even though `debug_plots=false`?

Adding a tab on that line to get `show` only called with all other `plt` code there fixes the issue, but it's more like a workaround, since the issue still persists if `Display sync plots` is enabled

artem-alexandrov, 2021-08-12:
Update 2:
Adding `block=False` as a parameter to `plt.show()` fixes the issue too

ElvinC, 2021-08-12:
Huh, thanks for the info. I made some changes to show all the plots at the same time (both sync and others) by only using one `plt.show` which works for me on windows.  The sync system will go through some changes to allow for a fully automatic sync and more than two points, so plots aren't final.

Just to be clear, does `block=False` affect the problem with persistent dropdown menus?

ElvinC, 2021-08-13:
In theory it's fixed

artem-alexandrov, 2021-08-13:
@ElvinC this one still happening when pressing "Export" button, I think there should be more `block=False` usages

ElvinC, 2021-08-13:
Thanks for the notice, with `block=False` I'm also getting some weirdness with some plots not appearing. Will try some different options.

ElvinC, 2021-08-22:
Added a check for OS, since the UI behavior seems to be different. This should be fixed now

closed 2021-08-22 by ElvinC