ElvinC/gyroflow #79: sync problem & QCoreApplication

minzukas, 2021-08-24:
Hi, 

I am still struglling to get it to work. My stabilized videos look actually worse than originals. I gues it is due to offset differences, not sure why though as I use video and csv straight out of runcam 5 orange, lens model is 1440 what I shoot with.  Perhaps anyone could help me out. 
ALso I get different graphs than shon on the totorial, but perhaps it is intended.
Tried versions 0.2.1 & 0.2.2 - get same results
[image]

[image]

ElvinC, 2021-08-24:
Did you try using auto sync in the current dev branch?

minzukas, 2021-08-24:
I get same shaky video if timestamps left unchanged in current dev

ElvinC, 2021-11-07:
Any luck since with the newer versions? Have a look here if not: https://docs.google.com/document/d/1mLrMd8itLYiGNdQh4oWqz-s3g16bfu-TkQNCS_ZqOnA/edit#

minzukas, 2021-11-07:
All good with newer versions, only wish non-linear stretching was working. Thanks a lot for

closed 2021-11-07 by minzukas