ElvinC/gyroflow #82: Auto installer installs an FFmpeg without libx264 support

Cam000, 2021-08-30:
The helper.py file in videogear.gear has a fixed URL for downloading FFmpeg in the absence of any binary on a Windows system. 

When it downloads this version during the video export there is a warning that it does not work with libx264, it then tries to use libxvid and fails (probably because it's still using the libx264 params).

To confirm this I changed the URL to a version that definitely has libx264 and it successfully rendered:

**helper.py, line 790**

```
file_url = "https://github.com/abhiTronix/FFmpeg-Builds/releases/latest/download/ffmpeg-static-{}-gpl.zip".format(
    os_bit
)
```
to

`file_url = "https://www.gyan.dev/ffmpeg/builds/ffmpeg-release-essentials.zip"`

ElvinC, 2021-08-30:
yes, that was the reason for recommending manually installing the ffmpeg version you need either in the same folder or the PATH. The vidgear auto download also by default places it in the temp folder, where it can be randomly deleted between use.

ElvinC, 2021-08-30:
Since ffmpeg needs to be manually installed on linux and mac, i'll change the message type back to error from your PR.

closed 2021-08-30 by ElvinC

abhiTronix, 2021-09-03:
@ElvinC Is it not working with `libx264` encoder?

abhiTronix, 2021-09-03:
@Cam000 I've just released new FFmpeg version with latest patches from upstream, which out-of-the-box supports following encoders. Hope that will solve your issue.

```sh
PS D:\> C:\Users\foo\AppData\Local\Temp\ffmpeg-static-win64-gpl/bin/ffmpeg.exe -encoders
ffmpeg version N-103482-g11d39873ab-20210903 Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 10-win32 (GCC) 20210408
  configuration: --prefix=/ffbuild/prefix --pkg-config-flags=--static --pkg-config=pkg-config --cross-prefix=x86_64-w64-mingw32- --arch=x86_64 --target-os=mingw32 --enable-gpl --enable-version3 --disable-debug --disable-w32threads --enable-pthreads --enable-iconv --enable-libxml2 --enable-zlib --enable-libfreetype --enable-libfribidi --enable-gmp --enable-lzma --enable-fontconfig --enable-libvorbis --enable-opencl --enable-libvmaf --enable-vulkan --disable-libxcb --disable-xlib --enable-amf --enable-libaom --enable-avisynth --enable-libdav1d --enable-libdavs2 --disable-libfdk-aac --enable-ffnvcodec --enable-cuda-llvm --enable-frei0r --enable-libglslang --enable-libgme --enable-libass --enable-libbluray --enable-libmp3lame --enable-libopus --enable-libtheora --enable-libvpx --enable-libwebp --enable-lv2 --enable-libmfx --enable-libopencore-amrnb --enable-libopencore-amrwb --enable-libopenjpeg --enable-librav1e --enable-librubberband --enable-schannel --enable-sdl2 --enable-libsoxr --enable-libsrt --enable-libsvtav1 --enable-libtwolame --enable-libuavs3d --disable-libdrm --disable-vaapi --enable-libvidstab --enable-libx264 --enable-libx265 --enable-libxavs2 --enable-libxvid --enable-libzimg --enable-libzvbi --extra-cflags=-DLIBTWOLAME_STATIC --extra-cxxflags= --extra-ldflags=-pthread --extra-ldexeflags= --extra-libs=-lgomp --extra-version=20210903
  libavutil      57.  4.101 / 57.  4.101
  libavcodec     59.  7.100 / 59.  7.100
  libavformat    59.  5.100 / 59.  5.100
  libavdevice    59.  0.101 / 59.  0.101
  libavfilter     8.  7.100 /  8.  7.100
  libswscale      6.  0.100 /  6.  0.100
  libswresample   4.  0.100 /  4.  0.100
  libpostproc    56.  0.100 / 56.  0.100
Encoders:
 V..... = Video
 A..... = Audio
 S..... = Subtitle
 .F.... = Frame-level multithreading
 ..S... = Slice-level multithreading
 ...X.. = Codec is experimental
 ....B. = Supports draw_horiz_band
 .....D = Supports direct rendering method 1
 ------
 V....D a64multi             Multicolor charset for Commodore 64 (codec a64_multi)
 V....D a64multi5            Multicolor charset for Commodore 64, extended with 5th color (colram) (codec a64_multi5)
 V..... alias_pix            Alias/Wavefront PIX image
 V..... amv                  AMV Video
 V....D apng                 APNG (Animated Portable Network Graphics) image
 V..... asv1                 ASUS V1
 V..... asv2                 ASUS V2
 V....D libaom-av1           libaom AV1 (codec av1)
 V....D librav1e             librav1e AV1 (codec av1)
 V..... libsvtav1            SVT-AV1(Scalable Video Technology for AV1) encoder (codec av1)
 V....D avrp                 Avid 1:1 10-bit RGB Packer
 V....D libxavs2             libxavs2 AVS2-P2/IEEE1857.4 (codec avs2)
 V..X.D avui                 Avid Meridien Uncompressed
 V....D ayuv                 Uncompressed packed MS 4:4:4:4
 V....D bmp                  BMP (Windows and OS/2 bitmap)
 VF.... cfhd                 GoPro CineForm HD
 V..... cinepak              Cinepak
 V....D cljr                 Cirrus Logic AccuPak
 V.S..D vc2                  SMPTE VC-2 (codec dirac)
 VFS..D dnxhd                VC3/DNxHD
 V....D dpx                  DPX (Digital Picture Exchange) image
 VFS..D dvvideo              DV (Digital Video)
 VF...D exr                  OpenEXR image
 V.S... ffv1                 FFmpeg video codec #1
 VF.... ffvhuff              Huffyuv FFmpeg variant
 V....D fits                 Flexible Image Transport System
 V..... flashsv              Flash Screen Video
 V..... flashsv2             Flash Screen Video Version 2
 V..... flv                  FLV / Sorenson Spark / Sorenson H.263 (Flash Video) (codec flv1)
 V..... gif                  GIF (Graphics Interchange Format)
 V..... h261                 H.261
 V..... h263                 H.263 / H.263-1996
 V.S... h263p                H.263+ / H.263-1998 / H.263 version 2
 V....D libx264              libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (codec h264)
 V....D libx264rgb           libx264 H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 RGB (codec h264)
 V....D h264_amf             AMD AMF H.264 Encoder (codec h264)
 V....D h264_mf              H264 via MediaFoundation (codec h264)
 V....D h264_nvenc           NVIDIA NVENC H.264 encoder (codec h264)
 V..... h264_qsv             H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10 (Intel Quick Sync Video acceleration) (codec h264)
 V....D libx265              libx265 H.265 / HEVC (codec hevc)
 V....D hevc_amf             AMD AMF HEVC encoder (codec hevc)
 V....D hevc_mf              HEVC via MediaFoundation (codec hevc)
 V....D hevc_nvenc           NVIDIA NVENC hevc encoder (codec hevc)
 V..... hevc_qsv             HEVC (Intel Quick Sync Video acceleration) (codec hevc)
 VF.... huffyuv              Huffyuv / HuffYUV
[... 131 lines cut ...]
 A....D real_144             RealAudio 1.0 (14.4K) (codec ra_144)
 A....D roq_dpcm             id RoQ DPCM
 A..X.D s302m                SMPTE 302M
 A....D sbc                  SBC (low-complexity subband codec)
 A..X.. sonic                Sonic
 A..X.. sonicls              Sonic lossless
[... cut ...]
 S..... ass                  ASS (Advanced SubStation Alpha) subtitle
 S..... dvbsub               DVB subtitles (codec dvb_subtitle)
 S..... dvdsub               DVD subtitles (codec dvd_subtitle)
 S..... mov_text             3GPP Timed Text subtitle
 S..... srt                  SubRip subtitle (codec subrip)
 S..... subrip               SubRip subtitle
 S..... text                 Raw text subtitle
 S..... ttml                 TTML subtitle
 S..... webvtt               WebVTT subtitle
 S..... xsub                 DivX subtitles (XSUB)
```

ElvinC, 2021-09-03:
> @ElvinC Is it not working with `libx264` encoder?
> 
> @ElvinC Also, new `v0.2.2` is out. Now you can remove all useless dependencies from your project that you needed with WriteGear. I find your work very promising, glad vidgear can help. Goodluck!

That's excellent, thanks for your work!