ElvinC/gyroflow #83: Gopro6 export rendering error

opendraftfpv, 2021-08-31:
I am very interested in gyroflow and am doing various tests.
I watched the runcam 5 orange video and was impressed.
However, the following error occurs when rendering Gopro6 export.
I tested the file with successful processing in reelsteady go.
My friends using gopro6 are also having the same problem.
Help is needed.
==========================================================
Rendering:   0%|?                                                                 | 56/16662 [00:20<1:27:52,  3.15it/s]fRendering:   0%|?                                                                 | 62/16662 [00:23<1:41:17,  2.73it/s]Unable to read multiple frames
Rendering:   0%|?                                                                 | 62/16662 [00:23<1:45:11,  2.63it/s]
Render finished
22:23:35 :: WriteGear :: DEBUG :: Terminating WriteGear Processes.
frame=   57 fps=2.4 q=-1.0 Lsize=    9765kB time=00:00:00.90 bitrate=88790.0kbits/s speed=0.0374x
video:9763kB audio:0kB subtitle:0kB other streams:0kB global headers:0kB muxing overhead: 0.016234%
[libx264 @ 0000026b856e5500] frame I:1     Avg QP:10.10  size:1129720
[libx264 @ 0000026b856e5500] frame P:14    Avg QP:13.14  size:453953

ElvinC, 2021-08-31:
Sometimes there are a couple frames right at the beginning of GoPro files which can't be read for some reason. Try setting the render range to start a few seconds later.

opendraftfpv, 2021-08-31:
Following your guide, the problem was solved. :)
In reelsteady go, the vibration was not resolved even if I adjusted the sync point.
In gyroflow, the processed result is very good.
Thank you so much.

closed 2021-08-31 by ElvinC