ElvinC/gyroflow #84: Allow use Oviedo split into multiple files.

avsaase, 2021-08-31:
FAT32 drives have a max file size of 4GB, so when the video file hits that limit Rican cameras stop recording and immediately start recording a new file. This causes a los of a second of video. To circumvent this I use loop recording to limit the videos to 5 minutes. With this setting there is no notacible gap between the two recordings but obviously you end up with multiple files even for shorter flights. 

It would be nice if Gyroflow could import multiple video files and paste them together at render time. Right now I merge the videos in VideoProc (which is quite fast and without quality loss because it doesn't have to re-encode the video) before processing in Gyroflow, but it still is an extra step. I would also not be surprised if there are some missed frames between the files, and if this causes problems with the sync. When piecewise sync is implemented then sync points could be placed around the file breakpoints to correct for any missed frames.

bakwc, 2023-03-04:
Same here with DJI O3 - it splits video into multiple file, need to merge somehow to stabilize together. When stabilize separately - there is a noticeable lag between files.

ElvinC, 2023-03-04:
@bakwc This repo is for the inactive python version. Please find the project here: https://github.com/gyroflow/gyroflow

closed 2023-03-04 by avsaase