ElvinC/gyroflow #87: Out of memory killer

brousselle, 2021-09-03:
Hello,
tried this new version on ubuntu 20.04 with git tag 0.3.0-beta
Everything went smoothly until rendering.
I have a laptop with 16G of RAM, and while rendering process gyroflow.py eats more and more RAM until it gets killed by OutOfMemoryKiller.

It was not doing so on previous versions.
Thanks :D

Pelado-Mat, 2021-09-06:
I can confirm the same behavior with Ubuntu 20.04.3 LTS. gyroflow.py eats my 26G of RAM and 2gb of swap.

ElvinC, 2021-09-06:
Thanks for the info. I actually had the same problem on my VM but figured it was just the 512 MB of ram causing issues. I'll try looking into it, but if anyone finds the cause, let me know

Pelado-Mat, 2021-09-06:
Just a quick test, disabling "Display preview during rendering" keeps the memory under control.

ElvinC, 2021-09-06:
Oh interesting. Then I have an idea of what's happening. I made some changes to the preview to show the frames in the main viewer instead of a popup window. This works on windows and mac, but when I tried in the ubuntu VM, the viewer didn't update with preview frames. Didn't think much of this, but it could be that all the frames piped to the viewer are queued in ram while waiting for something.