ElvinC/gyroflow #89: Use your phone's gyroscope for video stabilization

mppcasc, 2021-09-07:
Thank you for sharing. I now strap my camera to the back of my phone and use the information from the phone's gyroscope to stabilize the images taken by the camera.I got a bad result on your project.
I noticed in your project that the **GyroIntegrator** class has the following line of code "self.data[:, 2] *= -1". Why is this?
Also,When you call the get_maps function to output the video, there is a code "R = Rotation([-quat[1], -quat[2], quat[3], -quat[0]]).as_matrix()". Does this relate to the previous step?

Hope to get your reply, thank you.

ElvinC, 2021-09-09:
That is no longer part of the codebase, since it was modified to make better mathematical sense

mppcasc, 2021-09-10:
Thank you. I saw the modifications in **Gyroflow 0.3.0-beta**.
I also have one question:
Gyroscope data [x, y, z], the image rotation given by optical flow method is [-x, y, z], is this because the camera coordinate system is different from the image coordinate system? For example: The camera coordinate system is right-handed and the image coordinate system is left-handed? Can I understand it this way?

mppcasc, 2021-09-10:
Also, is the gyroscope sampling rate of 50Hz not enough? If not, what is the minimum sampling rate for 30fps video?

ElvinC, 2021-09-10:
> is this because the camera coordinate system is different from the image coordinate system

That's exactly it

> is the gyroscope sampling rate of 50Hz not enough

From what I've seen, the camera movements are usually around 0 to 40 Hz. Using nyquist with some margin gives 100 Hz as the minimum reasonable logging rate, if no aliasing is present.