ElvinC/gyroflow #90: The app crashes if moving the play head (left panel) a second time in a row, before the video frame upgrades the view.

dentex, 2021-09-07:
I hope it's clear from the title :)

```
Analyzing frame: 100%|███████████████████████████████████████████████████| 60/60 [00:36<00:00,  1.65it/s]
Estimated offset: 0.030000000000000027
Better offset: 0.01660000000000003, cost: 0.8123961549948631
98
3102
84
2659
76
2405
25.0 degree gyro/camera angle
Analyzing frame: 100%|███████████████████████████████████████████████████| 60/60 [00:42<00:00,  1.42it/s]
Estimated offset: 0.050000000000000044
Better offset: 0.04440000000000005, cost: 1.0701237180417293
146
4621
147
4653
Assertion fctx->async_lock failed at libavcodec/pthread_frame.c:155
Annullato (core dump creato)
```

ElvinC, 2021-09-09:
Thanks for the bug report, will look into it.

ElvinC, 2021-11-07:
Issue was due to conflict between two threads. Should be fixed in the dev branch now

closed 2021-11-07 by ElvinC