ElvinC/gyroflow #91: Export with Audio failing

deadbytefpv, 2021-09-08:
I'm having issues exporting with audio intact.
I'm getting the following error in the Log:
[Imgur](https://i.imgur.com/BJKTT73.jpg)

The drive where I set my output files to have enough disk space (I'm only stabilizing ~2GB mp4 files from a runcam 5 orange).

ElvinC, 2021-09-09:
Some fixes to the audio export have been made on this branch: https://github.com/Gro2mi/gyroflow/tree/auto_sync_point_suggestion

deadbytefpv, 2021-09-10:
> Some fixes to the audio export have been made on this branch: https://github.com/Gro2mi/gyroflow/tree/auto_sync_point_suggestion

Thanks. Although I'm having issues getting that branch running on my local copy..

deadbytefpv, 2021-09-11:
I think I may have found one major reason my Audio Exports aren't working.

I tried encoding the audio with the ffmpeg command in the stablizer.py script

`ffmpeg -i RC_0003_210905085602.mp4 -vn -acodec copy audio.mp4`

and got this:

`Could not find tag for codec pcm_s16le in stream #0, codec not currently supported in container`

Video clip is a Runcam 5 Orange clip. 

I tried to encode the audio with AAC and it went through.. 

`ffmpeg -i RC_0003_210905085602.mp4 -vn -acodec copy -c:a aac audio.mp4`

ElvinC, 2021-11-15:
Fixed in merged PR

closed 2021-11-15 by ElvinC