ElvinC/gyroflow #92: non-linear strech error

minzukas, 2021-09-08:
selecting non-linear stretch gives:

qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem

ElvinC, 2021-09-09:
Thanks for the bug report, could you give some info about which system/OS you're running?

minzukas, 2021-09-10:
Sure, i'm on ubuntu 20.04 lts!

minzukas, 2021-09-12:
some updates:
non-linear stretch still works on 0.2.2. beta, however only first time after restarting the pc. If I close video rendering screen and try to open new file and save it in new res I get this on terminal.
Closing now
Traceback (most recent call last):
  File "gyroflow.py", line 142, in open_stretch_util
    if self.calibrator_utility.isVisible():
AttributeError: 'NoneType' object has no attribute 'isVisible'

If I close terminal and restart .py exporting just crashes program altogether with the terminal.