ElvinC/gyroflow #93: Video only renders first second or so with GoPro Hero 8 2.7k 30 and 60 fps on MacOS

christopher-kapic, 2021-09-09:
I'm not sure if this is related to MiniGod's [issue](https://github.com/ElvinC/gyroflow/issues/57)--everything seems to work fine, until I actually try to export the video. When I check the result, only the first second or so is rendered, and then I get a black screen with just audio for the remainder of the video.

ElvinC, 2021-09-09:
Please add the relevant errors etc. that appear.

christopher-kapic, 2021-09-09:
I got it to work, but I don't recall what I did. If it happens again I will submit a new issue. For now, I will close it. Sorry to bother you.

closed 2021-09-09 by christopher-kapic

ElvinC, 2021-09-09:
Could the the third question listed here: https://elvinchen.com/gyroflow/FAQ/

christopher-kapic, 2021-09-10:
I think that's it.