ElvinC/gyroflow #105: Encoder h264_nvenc issue

jesuslg123, 2021-10-03:
Hello,

I am running Gyroflow 0.3.0-beta on Windows 10 with nvidia 3060 and when using the encoder h264_nvenc it completes the render but it shows error and only VLC can play the video.

1. Load video
2. Load gyroscope information from blackbox CSV
3. Autosync
4. Select any stabilization
5. Choose video encode h264_nvenc
6. Export

Render will complete with error.

`Rendering:  99%|███████████████████████████████████████████████████████████████▍| 6521/6575 [05:01<00:02, 21.64it/s]
Render finished
Traceback (most recent call last):
  File "gyroflow.py", line 2770, in export_video
  File "stabilizer.py", line 1569, in renderfile
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'C:/Users/jesus/Temporal/Oficina 7planta/flow/3D.mp4_a.mp4' -> 'C:/Users/jesus/Temporal/Oficina 7planta/flow/3D.mp4'`

The video can be played with VLC but not with Windows media player.

Thank you.

ElvinC, 2021-11-07:
Last couple of pull requests fixed some audio export issues. You can see if the dev branch works now

jesuslg123, 2021-12-05:
Hi @ElvinC ,

I just did an stabilization using h254_nvec and it is working. I have nvidia graphic card but I still see the load on CPU instead of the GPU while rendering the video with h265_nvec encoder, is that ok?

Thanks!

ElvinC, 2021-12-05:
That's expected, only the encoding is on the GPU while some of the processing is still done on the CPU.

closed 2021-12-05 by ElvinC