ElvinC/gyroflow #106: Error computing optimal flow

Alastorru, 2021-10-19:
Trying to stablilyzed insta360 go 2 video with default settings< and get the following error (trying with different files< the same result):
```
Output file: G:/YandexDisk/Media/Video/Insta360/210917/test.mp4
Using ffmpeg path C:\FPV\gyroflow
19:43:58 :: WriteGear :: DEBUG :: Compression Mode is enabled therefore checking for valid FFmpeg executables.
19:43:58 :: WriteGear :: DEBUG :: Output Parameters: {'-input_framerate': 29.97002997002997, '-vcodec': 'libx264', '-profile:v': 'high', '-crf': '1', '-maxrate': '82.0M', '-bufsize': '98M', '-pix_fmt': 'yuv420p'}
19:43:58 :: Helper :: DEBUG :: Final FFmpeg Path: C:\FPV\gyroflow\ffmpeg.exe
19:43:58 :: Helper :: DEBUG :: FFmpeg validity Test Passed!
19:43:58 :: Helper :: DEBUG :: Found valid FFmpeg Version: `b'2021-10-18-git-d04c005021-full_build-www.gyan.dev'` installed on this system
19:43:58 :: WriteGear :: DEBUG :: Found valid FFmpeg executables: `C:\FPV\gyroflow\ffmpeg.exe`.
19:43:58 :: WriteGear :: DEBUG :: Compression Mode is configured properly!
Starting to compute optimal Fov
adaptive_zoom.py:71: RuntimeWarning: invalid value encountered in true_divide
Traceback (most recent call last):
  File "gyroflow.py", line 2770, in export_video
  File "stabilizer.py", line 1339, in renderfile
  File "adaptive_zoom.py", line 123, in compute
  File "adaptive_zoom.py", line 123, in <listcomp>
  File "adaptive_zoom.py", line 75, in findFov
  File "scipy\interpolate\interpolate.py", line 544, in __init__
ValueError: `x` array is all-na
```
Last gyroflow executable, running on Windows10

Screens with settings are attached 
[image]
[image]
[image]
[image]
n

ElvinC, 2021-11-07:
hmm. Could you provide the files that gave the error?

KonradIT, 2021-11-07:
Sample from GO2: https://mega.nz/file/NsNCAJaQ#tJFbUnBXCqTrVwVxW7NFcso7KZ4ICagjMJmFX67w-_w

Alastorru, 2021-11-16:
My files:
https://disk.yandex.ru/d/eJIiG4WdGA-nFg
https://disk.yandex.ru/d/Lh-CQeglMbl71g
https://disk.yandex.ru/d/WPW9_wJ-p4f_IA

ElvinC, 2021-12-12:
Just tried one of the clips and it renders for me with `Insta360/Insta360_Go2_2880x2880_ProMode_unprocessed.json` on the current dev branch, so not sure what was causing the error. Setting the render in/out points to see if that helps.

Note that the gyro orientation might still be messed up since there isn't one specially for the go2. Maybe someone could record a handheld clip with some standalone roll pitch yaw movements?

KonradIT, 2021-12-14:
Here is a sample video from the go2 in pro video mode: https://mega.nz/file/Es8ETLIT#1xY1aFHIPwL1Y2IUWajIAq6WLt24_yj7cjU2_50Epjg

Alastorru, 2021-12-16:
> Just tried one of the clips and it renders for me with `Insta360/Insta360_Go2_2880x2880_ProMode_unprocessed.json` on the current dev branch, so not sure what was causing the error. Setting the [...]

Trying dev branch with settins You've mentioned. Works fine, optimal fov calculating error does not appear. Seems the problem was somehow fixed

Alastorru, 2021-12-20:
> Note that the gyro orientation might still be messed up since there isn't one specially for the go2. Maybe someone could record a handheld clip with some standalone roll pitch yaw movements?
Hi, Elvin. I've made 3 video for you: https://disk.yandex.ru/d/nCCtDNic_T83fw