ElvinC/gyroflow #112: poetry install fails on dev branch with darkdetect dependency error

SpoddyCoder, 2021-11-14:
Not a python dev, so this may be environment specific. Running on Windows 10, using gitbash...

```
$ poetry install
Creating virtualenv gyroflow-_9deXumQ-py3.9 in C:\Users\me_\AppData\Local\pypoetry\Cache\virtualenvs
Installing dependencies from lock file
Warning: The lock file is not up to date with the latest changes in pyproject.toml. You may be getting outdated dependencies. Run update to update them.

  SolverProblemError

  Because gyroflow depends on darkdetect (^0.5.0) which doesn't match any versions, version solving failed.

  at ~\.poetry\lib\poetry\puzzle\solver.py:241 in _solve
      237│             packages = result.packages
      238│         except OverrideNeeded as e:
      239│             return self.solve_in_compatibility_mode(e.overrides, use_latest=use_latest)
      240│         except SolveFailure as e:
    → 241│             raise SolverProblemError(e)
      242│
      243│         results = dict(
      244│             depth_first_search(
      245│                 PackageNode(self._package, packages), aggregate_package_nodes
```

Installed darkdetect with `pip install darkdetect` - completed OK - but still get the error above.

Master branch builds OK on same environment.

ElvinC, 2021-11-15:
Made some quick fixes to the dependencies, I think it should work now

SpoddyCoder, 2021-11-15:
Thanks for looking, I still got the same issue, but showing the new version pin. 

However, think I may have figured it out - looks like the `poetry.lock` file is out of step with the dependencies defined in `pyproject.toml`. Deleting the committed `poetry.lock` and running `poetry install` forces the deps to be re-resolved and the install sails through :)

To save you a bit of work, here's the list of dependencies that need to up be updated in the lock file...

```
darkdetect
ffprobe-python
qdarkstyle
pandas
vidgear
```

ElvinC, 2021-11-15:
Thanks for the info, I updated that as well, which should fix it now

closed 2021-11-15 by ElvinC