ElvinC/gyroflow #113: Dev branch CLI feature - sync error

SpoddyCoder, 2021-11-15:
Testing out @MiniGod 's CLI feature on the dev branch (https://github.com/ElvinC/gyroflow/pull/104). 

Using DJI Air Unit source video file and betaflight CSV. Tried a number of invocations, but output of minimal example below - error is during the sync phase, which goes onto repeat forever (only included the a small part of the 2nd repeat).

```
video.mp4 video.gyroflow.mp4 0.2 5
video.csv has log video.csv with type 'Blackbox CSV file'
Preset name: DJI _DJI FPV Air Unit__1080p 4by3
Note: Crop 1120x630 with 2x scale for calm flying
Made with 23 frames using calibrator version 0.2.1-alpha on date 2021-02-12
Note: Versions don't match. Calibrator: 0.3.0-beta-dev, Preset: 0.2.1-alpha. Should be fine though.
video.csv
Blackbox CSV file
Blackbox CSV file
average_frame_duration: 0.0005000000601974476
missing_runs: []
Appending 59.17091..=92.394914
average_frame_duration: 0.0005000000601974476
missing_runs: []
Appending 59.17091..=92.394914
2
Starting parallel auto sync
video.mp4 video.gyroflow.mp4video.mp4  0.2video.gyroflow.mp4  50.2 5

video.csv has log video.csv with type 'Blackbox CSV file'
video.csv has log video.csv with type 'Blackbox CSV file'
Preset name: DJI _DJI FPV Air Unit__1080p 4by3
Note: Crop 1120x630 with 2x scale for calm flying
Made with 23 frames using calibrator version 0.2.1-alpha on date 2021-02-12
Note: Versions don't match. Calibrator: 0.3.0-beta-dev, Preset: 0.2.1-alpha. Should be fine though.
Preset name: DJI _DJI FPV Air Unit__1080p 4by3
Note: Crop 1120x630 with 2x scale for calm flying
Made with 23 frames using calibrator version 0.2.1-alpha on date 2021-02-12
Note: Versions don't match. Calibrator: 0.3.0-beta-dev, Preset: 0.2.1-alpha. Should be fine though.
video.csv
video.csv
Blackbox CSV file
Blackbox CSV file
Blackbox CSV file
Blackbox CSV file
average_frame_duration: 0.0005000000601974476
average_frame_duration: 0.0005000000601974476
missing_runs: []
missing_runs: []Appending 59.17091..=92.394914

Appending 59.17091..=92.394914
average_frame_duration: 0.0005000000601974476average_frame_duration: 0.0005000000601974476

missing_runs: []
Appending 59.17091..=92.394914missing_runs: []

Appending 59.17091..=92.394914
2
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "C:\Python39\lib\multiprocessing\spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "C:\Python39\lib\multiprocessing\spawn.py", line 125, in _main
    prepare(preparation_data)
  File "C:\Python39\lib\multiprocessing\spawn.py", line 236, in prepare
    _fixup_main_from_path(data['init_main_from_path'])
  File "C:\Python39\lib\multiprocessing\spawn.py", line 287, in _fixup_main_from_path
    main_content = runpy.run_path(main_path,
  File "C:\Python39\lib\runpy.py", line 268, in run_path
    return _run_module_code(code, init_globals, run_name,
  File "C:\Python39\lib\runpy.py", line 97, in _run_module_code
    _run_code(code, mod_globals, init_globals,
  File "C:\Python39\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "E:\Temp-Dev\GyroFlow\gyroflow\cli.py", line 122, in <module>
[... 98 lines cut ...]

video.csv has log video.csv with type 'Blackbox CSV file'
Preset name: DJI _DJI FPV Air Unit__1080p 4by3
Note: Crop 1120x630 with 2x scale for calm flying
Made with 23 frames using calibrator version 0.2.1-alpha on date 2021-02-12
Note: Versions don't match. Calibrator: 0.3.0-beta-dev, Preset: 0.2.1-alpha. Should be fine though.
video.csv
video.csv
Blackbox CSV file
Blackbox CSV file
average_frame_duration: 0.0005000000601974476
missing_runs: []
Appending 59.17091..=92.394914
average_frame_duration: 0.0005000000601974476
Blackbox CSV file
Blackbox CSV file
average_frame_duration: 0.0005000000601974476
missing_runs: []
Appending 59.17091..=92.394914
missing_runs: []
Appending 59.17091..=92.394914
average_frame_duration: 0.0005000000601974476
missing_runs: []
Appending 59.17091..=92.394914
```

FreddyX078, 2021-11-24:
Got the same error with footage from Sony A7S3. GUI does not produce this error.

tomstom, 2022-01-29:
I checked the cli.py and the following line causes the issue in my case:

success = stab.full_auto_sync_parallel(
        max_fitting_error = args.auto_sync_error_margin,
        max_points=args.auto_sync,
        n_frames=args.sync_analyze_frames,
        sync_points=sync_points,
        debug_plots=False
    )

I changed it to:
success = stab.full_auto_sync(max_fitting_error = args.auto_sync_error_margin,max_points=args.auto_sync,debug_plots=False)

and in the full_auto_sync function in stabalizer.py I changed the return to:
return self.multi_sync_fit(max_fitting_error = max_fitting_error,debug_plots=debug_plots) #otherwise the gui starts

But remark: I think with this changes only the auto sync will work...