ElvinC/gyroflow #117: Nice to have: LUT support

fdev31, 2021-11-28:
Many cameras have their .cube 3D LUT files available, for simple workflows it could be convenient to integrate some color grading gyroflow since it's almost "for free" using ffmpeg.

An option to select a .cube file and then add it to the ffmpeg pipeline would be great.
As an illustration, such video filter option does the job:

`-vf lut3d=LUTS/Insta360/ONE-R-LUT/1-INCH.CUBE:interp=tetrahedral`

seanmavley, 2021-11-28:
I think currently you have the option to write in your custom ffmpeg command, therefore you can add in the above params during export.

[image]

fdev31, 2021-11-28:
Indeed, I agree, but it is not so "accessible" in my opinion for the following reasons:
- not user friendly, you need some ffmpeg knowledge to use it
- it does not allow easy selection of the .cube file (while a better integration allows a dropdown with a list of built-in CLUT for the most common cameras)
- it's quite unclear which "settings above" will be cleared if this field is provided, and how to replicate the settings
   - this last problem could be mitigated if this field somehow "auto filled in" according to the current settings, but this adds another level of complexity