ElvinC/gyroflow #120: Issue during Export

ntoenis, 2021-12-08:
Hello all, Attempted an export with debugging on in hopes someone can help me identify what's going wrong.

`Output file: D:/Users/Nick/Videos/Drone/Dec8th2021/SecondRun/GF_RC_0000_211208152126_Debug.MP4
Exporting gyroflow data file
Exporting complementary filter orientations
Finished exporting
Using ffmpeg path D:\Users\Nick\Documents\GitHub\gyroflow
15:50:34 ::   WriteGear    ::  DEBUG  :: Compression Mode is enabled therefore checking for valid FFmpeg executable.
15:50:34 ::   WriteGear    ::  DEBUG  :: Output Parameters: {'-input_framerate': 60.0, '-vcodec': 'h264_nvenc', '-profile:v': 'high', '-rc:v': 'cbr', '-b:v': '42.0M', '-bufsize:v': '84M'}
15:50:34 ::     Helper     ::  DEBUG  :: Final FFmpeg Path: D:\Users\Nick\Documents\GitHub\gyroflow\ffmpeg.exe
15:50:34 ::     Helper     ::  DEBUG  :: FFmpeg validity Test Passed!
15:50:34 ::     Helper     ::  DEBUG  :: Found valid FFmpeg Version: `b'2021-12-06-git-ef00d40e32-full_build-www.gyan.dev'` installed on this system
15:50:34 ::   WriteGear    ::  DEBUG  :: Found valid FFmpeg executable: `D:\Users\Nick\Documents\GitHub\gyroflow\ffmpeg.exe`.
15:50:34 ::   WriteGear    ::  DEBUG  :: Compression Mode with FFmpeg backend is configured properly.
Starting to compute optimal Fov
Done computing optimal Fov
Traceback (most recent call last):
  File "gyroflow.py", line 2928, in export_video
    self.stab.renderfile(start_time, stop_time, filename[0], out_size = out_size,
  File "D:\Users\Nick\Documents\GitHub\gyroflow\stabilizer.py", line 1081, in renderfile
    frame_temp = (cv2.resize(frame, out_size, interpolation=cv2.INTER_LINEAR) * 0.0).astype(np.float64)
numpy.core._exceptions.MemoryError: Unable to allocate 84.4 MiB for an array with shape (1440, 2560, 3) and data type float64`

This is 1440p footage from a Runcam 5 Orange using the dev branch.

ElvinC, 2021-12-08:
What are your system specs? Maybe try disabling render preview

ntoenis, 2021-12-08:
OS: Win 21H2
CPU: i7-8750H
RAM: 16GB
Will test disabling previewing (forgot about that option) and share results

ntoenis, 2021-12-08:
Hmm well that didn't seem to change much but different line noted in the error. Do you think changing bitrate might affect anything? 

```Output file: D:/Users/Nick/Videos/Drone/Dec8th2021/SecondRun/GF_RC_0000_211208152126_GyroDev.MP4                                                                         Using ffmpeg path D:\Users\Nick\Documents\GitHub\gyroflow                                                                                                                16:26:40 ::   WriteGear    ::  DEBUG  :: Compression Mode is enabled therefore checking for valid FFmpeg executable.                                                     16:26:40 ::   WriteGear    ::  DEBUG  :: Output Parameters: {'-input_framerate': 60.0, '-vcodec': 'h264_nvenc', '-profile:v': 'high', '-rc:v': 'cbr', '-b:v': '42.0M', '-bufsize:v': '84M'}                                                                                                                                                       16:26:40 ::     Helper     ::  DEBUG  :: Final FFmpeg Path: D:\Users\Nick\Documents\GitHub\gyroflow\ffmpeg.exe                                                           16:26:40 ::     Helper     ::  DEBUG  :: FFmpeg validity Test Passed!                                                                                                    16:26:40 ::     Helper     ::  DEBUG  :: Found valid FFmpeg Version: `b'2021-12-06-git-ef00d40e32-full_build-www.gyan.dev'` installed on this system                     16:26:40 ::   WriteGear    ::  DEBUG  :: Found valid FFmpeg executable: `D:\Users\Nick\Documents\GitHub\gyroflow\ffmpeg.exe`.                                            16:26:40 ::   WriteGear    ::  DEBUG  :: Compression Mode with FFmpeg backend is configured properly.                                                                    Starting to compute optimal Fov                                                                                                                                          Done computing optimal Fov                                                                                                                                               Traceback (most recent call last):                                                                                                                                         File "gyroflow.py", line 2928, in export_video                                                                                                                             self.stab.renderfile(start_time, stop_time, filename[0], out_size = out_size,                                                                                          File "D:\Users\Nick\Documents\GitHub\gyroflow\stabilizer.py", line 1078, in renderfile                                                                                     frame_out = cv2.resize(frame, out_size, interpolation=cv2.INTER_LINEAR) * 0.0                                                                                        numpy.core._exceptions.MemoryError: Unable to allocate 84.4 MiB for an array with shape (1440, 2560, 3) and data type float64 ```

ElvinC, 2021-12-09:
Are you using a 64 bit python installation?

ntoenis, 2021-12-09:
Looks like my environmental path for python is set to a 32-bit install.

ntoenis, 2021-12-09:
Success! I believe I had a number of issues here. Running 64 bit python was the key but also running 3.10.1 was too far. Went back and install Python 3.9 64 bit (reinstalled poetry) and things are rendering now. Thanks a ton