ElvinC/gyroflow #121: Gyroflow and gopro 5 experience so far

ricardouk, 2021-12-10:
Hi and first of all thank you for the work you have put on this app.

Ive tried to stabilize gopro 5 2.7k 4:3 non stabilized video and so far video doesnt get stablized, ive tried both the new and old method on the 0.3.0 version, the video im trying was shot from a window car while driving, been following your guide on youtube but so far no luck.

Has anyone had any luck with the hero 5 2.7k 4:3?

ElvinC, 2021-12-10:
Please post screenshots of the sync plots and input/output videos if possible

ricardouk, 2021-12-10:
Hi ElvinC, thanks for your help.

Im new to this program so be patient :)

https://ibb.co/FDnqP8p
https://ibb.co/G5ww8vC
https://ibb.co/26pyYyD
https://ibb.co/rkK7vSc

ricardouk, 2021-12-10:
Available log types
Gyroflow IMU log
Blackbox CSV file
Blackbox raw file
Runcam CSV log
Insta360 IMU metadata
GoPro GPMF metadata
ffmpeg version 2021-12-06-git-ef00d40e32-essentials_build-www.gyan.dev Copyright (c) 2000-2021 the FFmpeg developers
  built with gcc 11.2.0 (Rev2, Built by MSYS2 project)
  configuration: --enable-gpl --enable-version3 --enable-static --disable-w32threads --disable-autodetect --enable-fontconfig --enable-iconv --enable-gnutls --enable-libxml2 --enable-gmp --enable-lzma --enable-zlib --enable-libsrt --enable-libssh --enable-libzmq --enable-avisynth --enable-sdl2 --enable-libwebp --enable-libx264 --enable-libx265 --enable-libxvid --enable-libaom --enable-libopenjpeg --enable-libvpx --enable-libass --enable-libfreetype --enable-libfribidi --enable-libvidstab --enable-libvmaf --enable-libzimg --enable-amf --enable-cuda-llvm --enable-cuvid --enable-ffnvcodec --enable-nvdec --enable-nvenc --enable-d3d11va --enable-dxva2 --enable-libmfx --enable-libgme --enable-libopenmpt --enable-libopencore-amrwb --enable-libmp3lame --enable-libtheora --enable-libvo-amrwbenc --enable-libgsm --enable-libopencore-amrnb --enable-libopus --enable-libspeex --enable-libvorbis --enable-librubberband
  libavutil      57. 10.101 / 57. 10.101
  libavcodec     59. 14.100 / 59. 14.100
  libavformat    59.  9.102 / 59.  9.102
  libavdevice    59.  0.101 / 59.  0.101
  libavfilter     8. 19.100 /  8. 19.100
  libswscale      6.  1.101 /  6.  1.101
  libswresample   4.  0.100 /  4.  0.100
  libpostproc    56.  0.100 / 56.  0.100
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[1]/avcC/AVCConfig/reserved[]] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[1]/avcC/AVCConfig/reserved[]] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[0]/esds/ES/decConfigDescr/reserved] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
C:/Users/Utilizador/Desktop/file.MP4 has log C:/Users/Utilizador/Desktop/file.MP4 with type 'GoPro GPMF metadata'
Automatically detected gyro log file: file.MP4
Note: Versions don't match. Calibrator: 0.3.0-beta, Preset: 0.2.1-alpha. Should be fine though.
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[1]/avcC/AVCConfig/reserved[]] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[1]/avcC/AVCConfig/reserved[]] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
[warn] [/atom[2]/movie/atom[4]/track/atom[2]/media/atom[2]/minf/atom[2]/stbl/atom[0]/stsd/sample_entry[0]/atom[0]/esds/ES/decConfigDescr/reserved] padding contents doesn't look normal (invalid pattern)
[warn] [/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[5]/track/atom[1]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
Accl rate: 194.01984992244388 Hz, should be close to 200 or 400 Hz
GoPro GPMF metadata
GoPro GPMF metadata
QCoreApplication::exec: The event loop is already running
0.0 degree gyro/camera angle
Preset name: GoPro_Hero 5_2.7K 4:3_2.7K 4:3 wide
Note:
Made with 22 frames using calibrator version 0.2.1-alpha on date 2021-02-17
Note: Versions don't match. Calibrator: 0.3.0-beta, Preset: 0.2.1-alpha. Should be fine though.
C:/Users/Utilizador/Desktop/file.MP4
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Delete '/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr/name' (too large)
[warn] [/atom[2]/movie/atom[3]/track/atom[3]/media/atom[1]/hdlr] [Autofix] Fix parser error: stop parser, add padding
[... cut ...]
Analyzing frame: 100%|█████████████████████████████████████████████████████████████████| 60/60 [00:59<00:00,  1.00it/s]
Estimated offset: -0.04999999999999982
Better offset: -0.06159999999999982, cost: 0.869907826143388
0.0 degree gyro/camera angle
[-0.01602444  0.0342346 ]
{0, 1, 2, 3, 4}
Finished computing
Updating smoothness setting...
Smoothness updated
Output file: C:/Users/Utilizador/Desktop/hjkhjk.mp4
Using ffmpeg path C:\portable\Gyroflow
Starting to compute optimal Fov
Done computing optimal Fov

ElvinC, 2021-12-10:
Hmm, it that should've been alright, though the older cameras could have issues with bad rolling shutter. Could you send the video file?

ricardouk, 2021-12-10:
Thanks, im off to work, do you have a email i can send it to?

ElvinC, 2021-12-10:
Easiest would be uploading it to wetransfer and putting a link here

ricardouk, 2021-12-10:
ive tried and it requires email of the sender and the receiver, ill use another service.

https://www.dropbox.com/s/sg8u0krxv0w1c1h/file.MP4?dl=0

ElvinC, 2021-12-10:
Does the hero 5 have inbuilt stabilization? Looks like some stabilization is already applied to the clip in pitch and yaw...

ricardouk, 2021-12-12:
Hi Elvin C, i had to record some videos again, altough i was sure stabilization was off ive recorded some more videos with stabilization disabled but gyroflow still fails, i tested the demo version of reelstready and it worked with both new and old videos, i noticed gyroflow displays this info whenever i open a video:

"Preset name: GoPro_Hero 5_2.7K 4:3_2.7K 4:3 wide
Note:
Made with 22 frames using calibrator version 0.2.1-alpha on date 2021-02-17
Note: Versions don't match. Calibrator: 0.3.0-beta, Preset: 0.2.1-alpha. Should be fine though."

and it selects hero 6 on the gyro source.

Anything i could do to help?

ElvinC, 2021-12-12:
hero 6 is just selected by default that needs to be changed to hero 5

ElvinC, 2021-12-12:
Do you perhaps have some simple handheld clips with some rotation that you could send?

ElvinC, 2021-12-13:
Okay I might've figured it out. If you run the latest dev version (fixed some bugs), sync the video, and change the smoothing method to "roll pitch yaw smoothing", set yaw and pitch smoothing to zero, and crank up the roll smoothing, you get this which seems to work. So it does seem like the first clip you sent has some smoothing applied already

https://user-images.githubusercontent.com/20195216/145735043-11a83d95-4c13-4ce8-85ca-14627dc348cd.MP4

ricardouk, 2021-12-13:
so i just did a fresh windows 10 install and after following your advice here's what i got

Original: https://streamable.com/tsnmjn

reelsteady : https://streamable.com/vpafyh

gyroflow: https://streamable.com/7hg1em

ive tried to use gyroflow on ubuntu but even after reading some comments here i couldnt get it to work, have you thought about doing a appimage self conained version? Thanks

ElvinC, 2021-12-13:
Is that the same or a different file than what you previously sent?

And what exactly is not working on ubuntu. Please be more specific