ElvinC/gyroflow #122: Combining multiple files (over 4GB/12GB Limit)

fremebondo, 2021-12-11:
Hi there, 

if working with split files (4GB limit or 12 with labs firmware), which is the best workflow to combine the files and have them processed correctly by Gyroflow?

ricardouk, 2021-12-12:
you can combine all of the files without reconverting them with avidemux...as long as video/audio/dimensions are the same on all videos

ricardouk, 2021-12-12:
Open avidemux
open first video file
on video and audio output select "copy" on output format format select mp4
go to file menu and select append to open the second video file, do it again to add a third or fourth file
go to file menu and select save..thats it

ElvinC, 2021-12-13:
4 GB files can be concatenated through the process described here: https://gopro.github.io/labs/control/chapters/
Not sure if the results work with gyroflow though since I haven't tried yet

fremebondo, 2021-12-13:
> 4 GB files can be concatenated through the process described here: https://gopro.github.io/labs/control/chapters/ Not sure if the results work with gyroflow though since I haven't tried yet

I did it, before posting this issue. The resulting file won't work with gyroflow.

fremebondo, 2021-12-13:
> Open avidemux open first video file on video and audio output select "copy" on output format format select mp4 go to file menu and select append to open the second video file, do it again to add a third or fourth file go to file menu and select save..thats it

Thanks, I'll try this solution asap

MiniGod, 2021-12-15:
> I did it, before posting this issue. The resulting file won't work with gyroflow.

Regarding 12GB chapters... Gyroflow can't read the gyro data from a MP4 file bigger than 4 GB, but you can use the MP4 for video and the LRV file for gyro.
If you only have the mp4 file, you can use ffmpeg to compress the mp4 file to less than 4GB with the gpmf data mapped (maybe you don't need to map video and audio, maybe only the gpmf data needs to be mapped), then use that as gyro, but full quality video as video.

Of course, if you have a long video that splits at 12GB, you'd still have the same issue.

fremebondo, 2022-03-31:
@MiniGod I havent tried your solution yet , did you?

MiniGod, 2022-03-31:
Yes I've done it with > 4GB files straight from camera (below 12GB).
I haven't used this version of Gyroflow in a while though. Using the new version https://github.com/gyroflow/gyroflow. It supports big files like yours.

fremebondo, 2022-03-31:
no chance to stich back 4GB files instead?

fremebondo, 2022-04-01:
I am not a ffmpeg expert, but I understood that GoPro Labs solution moves gpmd stream from stream 0:3 to 0:2 
Could this be the problem?

fremebondo, 2022-04-01:
by the way, method suggested by @ricardouk does not work either as avidemux ignores metadata stream

MiniGod, 2022-04-01:
I don't have experience with stitching split files.

I've only compressed a single up-to 12GB chapter file to become less than 4GB so that gyroflow v0 (this repo) could load it. To do that I did the same mapping of the streams as in the link, `-map 0:0 -map 0:1 -map 0:3`, which indeed makes the gpmf stream `0:2`, so that should not be the problem.

I assume the issue is with the 2nd command (`udtacopy`), which I do not have experience with.

fremebondo, 2022-04-01:
> I am not a ffmpeg expert, but I understood that GoPro Labs solution moves gpmd stream from stream 0:3 to 0:2, while 0:2 usually contains tmcd stream, which I understand contains time sync data, and ffmpeg is not able to copy this stream. Could this be the problem? @ElvinC , can you confirm that a missing tmcd stream can cause issues in stabilization?

@MiniGod , just added some info here

fremebondo, 2022-04-01:
`udtacopy` may be the responsible. I cannot however find any documentation about this executable.

fremebondo, 2022-04-03:
@ElvinC , could you suggest how can I check gyro data consistency between split and joined files? Loading 1st chapter gives a excellent result, while loading the stitched file does not. 
Gyro waveform shown in GyroFlow is similar in the two files so I suspect some kind of scaling is sync issue.

Again, what is strange is : result on 1st file are comparable between GyroFlow and ReelSteady, while only RS si capable of working correctly with the stitched file.

Do you think I should upload these clips somewhere so they can be used for testing?

wdim0, 2022-08-08:
I happened to solve it (at least for me) - I join 4 GB files to one big video file like this (on Linux):
```
ffmpeg -f concat -safe 0 -i list.txt -c copy -map 0:v -map 0:a -map 0:3 -copy_unknown joined.mp4
```
where content of **list.txt** is something like this:
```
file 'input1.mp4'
file 'input2.mp4'
file 'input3.mp4'
```
`-safe 0` is there to get rid of annoying restriction if you're using full paths in **list.txt** (relative paths are ok)
`-map 0:v -map 0:a -map 0:3 -copy_unknown` is necessary to force ffmpeg to copy gpmd (GoPro's gyro) data

Tested with GF v1.0.0-rc - it reads gyro data from files bigger than 4 GB ok.

The only issue is, that GF doesn't automatically detect lens profile after loading **joined.mp4** but this you can fix youself manually in GF (look what's detected when using GF with one of your original input files and choose the same manually after loading **joined.mp4**)

fremebondo, 2022-08-08:
Thanks @wdim0 .... honestly I got very good results with https://github.com/rubegartor/ReelSteady-Joiner which has its GUI and do not require writing file lists or shell cmds. I'll try your solution as well next time I will join chapters.

Klattrup, 2022-08-16:
I just wanted to share that I used @wdim0 's ffmpeg solution to combine my files to a single 36gb file that GyreFlow opened, processed and rendered flawlessly. It even got the lens-data correct. Thank you all. :)

bakwc, 2023-03-04:
How to merge DJI O3 files? ReelSteady-Joiner and this ffmpeg approach doesn't work (

slo-addy, 2023-04-19:
> Actually everything already works, just drag-and-drop 3 files at the same time.

This is the way.

slo-addy, 2023-04-19:
> > Actually everything already works, just drag-and-drop 3 files at the same time.
> 
> This is the way.

Link to the docs for this https://docs.gyroflow.xyz/app/getting-started/file-joiner

lesau, 2023-07-08:
Hi, I tried joining 3 video file chapters from the split video made by CadDx Walnut camera using GF 1.5.1. GF joined them successfully and also created a .gcsv with the same name as described as the video -- so exactly as per https://docs.gyroflow.xyz/app/getting-started/file-joiner. However, the video doesn't get stabilized even around the green labeled sync points. I have 4 red and 2 green labeled sync points and gyro data appears empty/flat in the middle part of the video that corresponds to the 2nd gcsv file. I do have all 3 mp4 and 3 gcsv files, which are part of the same video split by the camera.

In another test, each video stabilizes properly by itself without joining using the same stabilization settings. So, something is iffy with the joining. Any ideas? 

FOV 1.0, DEFAULT smoothing, smoothness 0.5-0.75, zooming speed 2.5, rolling shutter left at defaults see below.

Picture: 3 videos joined w/ missing gyro data for the 2nd video:
[image]

Picture: zoomed in one sync point in the 1st video when joined, but the video is shaking in the joined video
[image]

[image]
[image]

ElvinC, 2023-07-08:
Hi, please direct this to https://github.com/gyroflow/gyroflow. Furthermore, attaching some sample files would be useful.

I'll archive this repo since it's now inactive