ElvinC/gyroflow #127: Insta GO2 stabilization not working?

Fobos25, 2021-12-28:
I tried to stabilize video from Insta GO2 camera. Unfortunately, the picture was not stable. Please tell me, is this camera not supported yet? Is only the first version supported?

ElvinC, 2021-12-30:
yeah it uses a different gyro orientation so the insta360 go option doesn't work. You can try the telemetry_parser log option using the dev branch, otherwise it'll be added in the future

Alastorru, 2022-01-12:
When select telemetry_parser log option, got an error on pressing "Plot and analyze gyro data" or "Attempt auto sync":
`C:/Users/User/Videos/PRO_VID_1440p_50fps_standart.mp4 has log C:/Users/User/Videos/PRO_VID_1440p_50fps_standart.mp4 with type 'Insta360 IMU metadata'
Automatically detected gyro log file: PRO_VID_1440p_50fps_standart.mp4
Note: Versions don't match. Calibrator: 0.3.0-beta-dev, Preset: 0.3.0-beta. Should be fine though.
Camera:  Insta360
Model:  Insta360 GO 2
Traceback (most recent call last):
  File "gyroflow.py", line 2499, in input_gyro_plot
    success = temp_log_reader.extract_log(self.gyro_log_path)
  File "C:\FPV\GFlow\gyroflow-dev\gyrolog.py", line 317, in extract_log
    self.extracted = self.extract_log_internal(filename)
  File "C:\FPV\GFlow\gyroflow-dev\gyrolog.py", line 1457, in extract_log_internal
    data = np.array([(entry["timestamp_ms"],) + entry["gyro"] + entry["accl"] for entry in norm_imu])
  File "C:\FPV\GFlow\gyroflow-dev\gyrolog.py", line 1457, in <listcomp>
    data = np.array([(entry["timestamp_ms"],) + entry["gyro"] + entry["accl"] for entry in norm_imu])
TypeError: can only concatenate tuple (not "NoneType") to tuple`
When select Insta360 IMU metadata, this error is not appear