ElvinC/gyroflow #128: GyroLog support for Sony Cams

tomstom, 2021-12-30:
It would be wounderfull to see in a next version of Gyroflow a support for some Sony Cams eg. RX100, ZV1, RX0M2. I found the project: https://github.com/SK-Hardwired/xavc_rtmd2srt which already did the extraction of the gyro log, I already tested with the ZV1, that it is possible to use these logs, I convert them to a BetaFlight CSV Log (factor is -65.91499036324716 to get the right scaling). 
If you need some files for testing, I have enough :-)

ElvinC, 2021-12-30:
If you run the dev branch or the dev exe from here: https://elvinchen.com/gyroflow/download/ you can select the "telemetry_parser" log option which works with sony logs. Alternatively this tool converts the sony video files to the blackbox format: https://github.com/AdrianEddy/telemetry-parser

closed 2021-12-30 by ElvinC

hi6otb, 2022-01-10:
That's mean "using Eddy's telemetry-parser and select telemetry_parser on log option?"

tomstom, 2022-01-10:
The problem is, if you try to use "telemetry_parser" log option direct without conversation, the gyro values are not scaled right,  I think the division by the factor mentioned above is missing... Maybe an option for the zv1 next to the "default" in the combobox would be nice with correct sacking.

Ps: the factor for the zv1 is only calculated by an integration of a 90° rotation done by hand and then averaged over 6 measurements , so it is not the 100% correct value... but I think it's better then without scaling ;-)