ElvinC/gyroflow #132: MacOS Installation Guide?

stimfish, 2022-01-07:
I've tried downloading the zip dmg file for mac and unzipped it, paste chmod +x ./downloads/gyroflow to Terminal and run it. After which the gyroflow file became an executable file. However i have no idea how to run it. I tried double clicking on the gyroflow file and another Terminal window opened (as attached). I do not know how to proceed further and could not find any instructions on MacOS installation. Please help! Thank you
[image]
.

ElvinC, 2022-01-07:
did you try running just `./downloads/gyroflow`?

stimfish, 2022-01-08:
Hi ElvinC, I've ran ./downloads/gyroflow but the pop up says i need FFMPeg. I have downloaded FFMPeg but it is running as executable on a seperate Terminal window.
[image]

TimmysG, 2022-01-11:
Hey @stimfish! I had this exact same issue, here’s a super helpful video on how to fix it.
https://youtu.be/H1o6MWnmwpY

stimfish, 2022-01-25:
> Hey @stimfish! I had this exact same issue, here’s a super helpful video on how to fix it. https://youtu.be/H1o6MWnmwpY

Hi @TimmysG thank you so much for the guide! I have successfully installed ffmpeg. Now the issue is the lens preset which I have no idea where to find it. I have a Mobula6 HD which uses Runcam and Naked GoPro 8.

@ElvinC Any idea where can i find the lens preset? Thank you so much!