ElvinC/gyroflow #157: Black Orbs Visible Throughout footage

E5TIE, 2023-04-18:
Using a new Gopro Hero 11 mini. I'm getting weird black orbs throughout the footage. Check all settings to see if there's a mistake but it seems like the gyro data is not reading correctly. The same issue happen when using the slipt recording option.
[image]
[image]

E5TIE, 2023-04-18:
The IMU orientation seems to change after a flip. Does that mean that the orientation feature on the gopro has to be locked?

ElvinC, 2023-04-19:
This is the inactive legacy version. Please go here: https://github.com/gyroflow/gyroflow

closed 2023-04-19 by ElvinC